OpenXiangShan/XiangShan #30: logutils: 需要一些改进

poemonsense, 2020-06-24:
(1) 需要支持对特定时间范围内的log打印
实现上，只需要在XSlog里面cond加入条件即可

(2) 需要支持对特定模块log打印，在代码中唯一的地方选择部分模块进行打印，能够开关每一个模块的打印。
具体实现上，可以通过对模块的选择来做。

ljwljwljwljw, 2020-06-24:
我尽快加入这些功能

ljwljwljwljw, 2020-06-25:
#43 该PR中加入了指定log时间范围的功能， 通过在emu中加入-b和-e选项控制log的开始和结束时间。

ljwljwljwljw, 2020-06-25:
针对特定模块进行log控制的功能尚未实现

ljwljwljwljw, 2020-06-25:
#44 该PR中加入了在emu中指定log level的功能，只通过emu的-l参数就可以修改log level，不需要重新编译chisel到verilog

AugustusWillisWang, 2020-07-06:
需要添加在处理器状态错误时, 无视当前处理器设置输出所有调试信息的代码. 

在后期运行复杂的测试/操作系统时, 需要仿真很久才能触发bug. 这些测试的指令量很大, 一直生成log会由于IO瓶颈导致仿真缓慢. 而关闭log运行的话, 在测试发现问题时无法输出处理器的内部信息.

closed 2020-12-09 by ljwljwljwljw

biquanha, 2023-11-23:
[translate]：(1) You need to support log printing within a specific time range. In
terms of implementation, you only need to add conditions to cond in XSlog

(2) It needs to support the printing of specific module logs, select some modules in the only place in the code for printing, and be able to switch the printing of each module.
In terms of specific implementation, it can be done through the selection of modules.

biquanha, 2023-11-23:
[translate]:#43 This PR adds the ability to specify the log time range, and controls the start and end times of the log by adding -b and -e options to the emu.

biquanha, 2023-11-23:
[translate]:You need to add code that ignores the current processor settings and outputs all debug information when the processor state is wrong.

When running complex tests/operating systems at a later stage, it takes a long time to emit bugs. These tests have a large number of instructions, and the constant generation of logs will cause the simulation to be slow due to IO bottlenecks. If you turn off the log, you will not be able to output the internal information of the processor when the test finds a problem.