OpenXiangShan/XiangShan #49: 重命名表初始化与后续维护存在问题

poemonsense, 2020-06-26:
现在的重命名表全部初始化为0，同时所有物理寄存器均在freelist中，同时没有对0号寄存器做特殊处理。
这会导致在rename分配0号物理寄存器后，dispatch2阶段发现源操作数(初始为0号)是busy状态，等待写回，因此指令被一直堵在保留站中无法发射。

poemonsense, 2020-06-26:
为了运行第一条指令，暂时简单地从freelist中移除0号物理寄存器，问题仍然没有彻底解决

ljwljwljwljw, 2020-06-27:
重命名表已修复 #53

closed 2020-06-27 by AugustusWillisWang

biquanha, 2023-11-23:
[translate]:The rename table is now all initialized to 0, and all physical registers are in the freelist, and there is no special handling for register 0.
This causes the Dispatch0 phase to find that the source operand (initially 2) is in a busy state after the rename is assigned to physical register 0, waiting to be written back, so the command is stuck in the reserved station and cannot be transmitted.

biquanha, 2023-11-23:
[translate]:In order to run the first instruction, temporarily and simply remove the number 0 physical register from the freelist, the problem is still not completely solved

biquanha, 2023-11-23:
[translate]:Renaming tables has been fixed #53