OpenXiangShan/XiangShan #100: 对飞线的管理

ljwljwljwljw, 2020-07-09:
目前的代码中有许多使用BoringUtils进行飞线连接的地方，使用这些飞线的目的大致可以分为3类：
1.跨模块传递Debug信号
2.接性能计数器
3.简化模块间连线

由于BoringUtils并没有一个集中管理飞线的地方，许多使用了飞线连接的模块被删除/修改后难以发现错误，例如：
1.如果Source被删除，Sink不会报错，但Sink的值永远不会被Source更新
2.如果Sink被删除，代码编译会报错，如果代码中有多个Sink被删除，只能根据报错信息一个一个寻找，调试效率很低

基于上述原因，需要一个模块对所有飞线统一管理，并能提供API查询代码中存在的所有飞线

ljwljwljwljw, 2020-07-15:
#127 该PR加入了ExcitingUtils，对BoringUtils进行了封装，可以提供对飞线的查询和检查功能，之后需要加入根据飞线类型开启、关闭飞线的功能

closed 2020-12-09 by ljwljwljwljw

biquanha, 2023-11-23:
[translate]:There are many places in the current code that use BoringUtils for flying wire connection, and the purpose of using these flying wires can be roughly divided into three categories:
3. Transmit Debug signals across modules
1. Connect performance counters
2. Simplify the connection between modules

Since BoringUtils does not have a centralized place to manage flying cables, many modules that use flying cable connections are deleted/modified and difficult to find errors, such as:
1. If the Source is deleted, the Sink will not report an error, but the value of the Sink will never be updated by the Source
2. If multiple sinks are deleted, you can only search for them one by one based on the error messages, which is very inefficient

For the above reasons, a module is required to manage all flying wires in a unified manner and provide all flying wires in the API query code

biquanha, 2023-11-23:
[translate]:#127 This PR adds ExcitingUtils, which encapsulates BoringUtils, which can provide the function of querying and checking flying wires, and then need to add the function of turning on and off flying wires according to the type of flying wires