OpenXiangShan/XiangShan #337: 关于 `src/test/csrc/main.cpp` 里面返回值的问题

SKTT1Ryze, 2020-12-21:
目前 `src/test/csrc/main.cpp` 里面是这样返回的：  
```cpp
return !is_good_trap;
```
这样写的话 `emu` 返回值只能是 0 或 1, 返回 0 代表程序正常返回，仿真一切正常；返回 1 即表示程序异常返回，仿真中出现问题，但是我们不知道是什么异常。  
这样带来的问题：  
+ 监控程序或其他外部程序无法从返回结果知道仿真过程中出现什么类型的错误
+ `emu` 后带 `-I` 参数后即使仿真正确，返回值也是 `1`，这时候自动化测试脚本将会认为该仿真过程出现了错误，从而回退 `10000` 个周期继续仿真。虽然我们可以检查输出文件来判断这种带 `-I` 参数的仿真正常与否，但这不是一个优雅的做法。

个人建议做以下修改：  
```cpp
// emu.h:  
class Emulator {
 ......
endif
EmuArgs args;

enum {
 STATE_GOODTRAP = 0,
 STATE_BADTRAP = 1,
 STATE_ABORT = 2,
 STATE_LIMIT_EXCEEDED = 3,
 STATE_RUNNING = -1
 };

......

public:  
......
bool is_good_trap() { return trapCode == STATE_GOODTRAP; };
int trap_code() { return trapCode; }
};

// main.cpp:  
int main(int argc, const char** argv) {
......
int trapcode = emu->trap_code();
 ......
 return trapcode;
}
```

这样子我们就可以通过程序返回值来判断仿真的状态了，比如：  
```python
try:
            cmd( # TODO  )
        except sh.ErrorReturnCode_1 as e:
            # TODO
            pass
        except sh.ErrorReturnCode_2 as e:
            # TODO
            pass
        except sh.ErrorReturnCode_3 as e:
            # TODO
            pass
```

SKTT1Ryze, 2020-12-21:
close

closed 2020-12-21 by SKTT1Ryze

biquanha, 2023-11-23:
[translate]:At the moment it returns like this:src/test/csrc/main.cpp

return !is_good_trap;
If you write it this way, the return value can only be 0 or 1, returning 0 means that the program returns normally, and everything is fine in the simulation, and returning 1 means that the program returns abnormally, and there is a problem in the simulation, but we don't know what the exception is.
Problems with this:emu

Monitors or other external programs cannot tell what type of error occurred during the simulation from the results returned
emu Even if the simulation is correct with the parameters, the return value is still , at which point the automated test script will think that there is an error in the simulation process and will roll back one cycle to continue the simulation. While we can check the output file to determine if this parametric simulation is working or not, this is not an elegant approach.-I110000-I
Personally, I recommend making the following changes:

// emu.h:  
class Emulator {
 ......
endif
EmuArgs args;

enum {
 STATE_GOODTRAP = 0,
 STATE_BADTRAP = 1,
 STATE_ABORT = 2,
 STATE_LIMIT_EXCEEDED = 3,
 STATE_RUNNING = -1
 };

......

public:  
......
bool is_good_trap() { return trapCode == STATE_GOODTRAP; };
int trap_code() { return trapCode; }
};

// main.cpp:  
int main(int argc, const char** argv) {
......
int trapcode = emu->trap_code();
 ......
 return trapcode;
}
In this way, we can judge the state of the simulation by the return value of the program, for example:

try:
            cmd( # TODO  )
        except sh.ErrorReturnCode_1 as e:
            # TODO
            pass
        except sh.ErrorReturnCode_2 as e:
            # TODO
            pass
        except sh.ErrorReturnCode_3 as e:
            # TODO
            pass