OpenXiangShan/XiangShan #837: make verilog error

vitaminabcd, 2021-06-24:
我的环境是debian，按照说明git clone了本项目，然后执行make init，然后执行make verilog，报错如下：请赐教，感谢！
make verilog
mkdir -p build
mill XiangShan.test.runMain top.TopMain -td build --config DefaultConfig --full-stacktrace --output-file XSTop.v --disable-all --remove-assert --infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf
[104/212] rocket-chip.compile
[info] compiling 376 Scala sources to /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/out/rocket-chip/compile/dest/classes ...
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:293:25: not found: value hardfloat
[error]   def recode(x: UInt) = hardfloat.recFNFromFN(exp, sig, x)
[error]                         ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:294:23: not found: value hardfloat
[error]   def ieee(x: UInt) = hardfloat.fNFromRecFN(exp, sig, x)
[error]                       ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:465:25: not found: value hardfloat
[error]   val dcmp = Module(new hardfloat.CompareRecFN(maxExpWidth, maxSigWidth))
[error]                         ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:493:29: not found: value hardfloat
[error]       val conv = Module(new hardfloat.RecFNToIN(maxExpWidth, maxSigWidth, xLen))
[error]                             ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:503:35: not found: value hardfloat
[error]           val narrow = Module(new hardfloat.RecFNToIN(maxExpWidth, maxSigWidth, w))
[error]                                   ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:551:28: not found: value hardfloat
[error]       val i2f = Module(new hardfloat.INToRecFN(xLen, t.exp, t.sig))
[error]                            ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:560:55: value >> is not a member of Nothing
[error]     val dataPadded = data.init.map(d => Cat(data.last >> d.getWidth, d)) :+ data.last
[error]                                                       ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:560:60: value getWidth is not a member of Nothing
[error]     val dataPadded = data.init.map(d => Cat(data.last >> d.getWidth, d)) :+ data.last
[error]                                                            ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:614:35: not found: value hardfloat
[error]         val narrower = Module(new hardfloat.RecFNToRecFN(maxType.exp, maxType.sig, outType.exp, outType.sig))
[error]                                   ^
[error] /mnt/d/MyCoud/typoradocs/Tech/cpu/xiangshan/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:648:46: not found: value hardfloat
[error]     val mulAddRecFNToRaw_preMul = Module(new hardfloat.MulAddRecFNToRaw_preMul(expWidth, sigWidth))
[error]                                              ^
[error] 14 errors found
1 targets failed
rocket-chip.compile Compilation failed
make: *** [Makefile:52: build/XSTop.v] Error 1

poemonsense, 2021-06-24:
问题可能是网络环境导致submodule初始化失败导致的。

可以先尝试以下几个解决方案：
(1) 再次make init
(2) rm -rf out之后，重新make init
(3) 删除rocket-chip, berkeley-hardfloat等submodule目录，重新make init

如果仍然存在问题，可以试一下重新git clone XiangShan

vitaminabcd, 2021-06-24:
> 问题可能是网络环境导致submodule初始化失败导致的。
> 
> 可以先尝试以下几个解决方案：
> (1) 再次make init
> (2) rm -rf out之后，重新make init
> (3) 删除rocket-chip, berkeley-hardfloat等submodule目录，重新make init
> 
> 如果仍然存在问题，可以试一下重新git clone XiangShan

感谢，感谢！我再试试！

vimcdoe, 2021-06-25:
make verilog  
mkdir -p build
mill XiangShan.test.runMain top.TopMain -td build --config DefaultConfig --full-stacktrace --output-file XSTop.v --disable-all --remove-assert --infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf 
[212/212] XiangShan.test.runMain 
Elaborating design...
FPGASoC cores: 1 banks: 4 block size: 64 bus size: 256
[WARN] Signal |DISPLAY_LOG_ENABLE| has multiple sinks
[WARN] Signal |logTimestamp| has multiple sinks
ImmUnion max len: 20
0: exu:LoadExu fastPortsCnt: 8 slowPorts: 4 delay:-1 feedback:true
1: exu:LoadExu fastPortsCnt: 8 slowPorts: 4 delay:-1 feedback:true
2: exu:StoreExu fastPortsCnt: 6 slowPorts: 12 delay:-1 feedback:true
3: exu:StoreExu fastPortsCnt: 6 slowPorts: 12 delay:-1 feedback:true
LoadQueue: size:64
StoreQueue: size:48
FPGAPlatform:true EnableDebug:true

Memory map:
[h00_0000_0000 -> h00_0FFF_FFFF] Width:unlimited Description:Reserved [RW]
[h00_1000_0000 -> h00_1FFF_FFFF] Width:unlimited Description:QSPI_Flash [RWX]
[h00_2000_0000 -> h00_2FFF_FFFF] Width:unlimited Description:Reserved [RW]
[h00_3000_0000 -> h00_3000_FFFF] Width:unlimited Description:DMA [RW]
[h00_3001_0000 -> h00_3004_FFFF] Width:unlimited Description:GPU [RWC]
[h00_3005_0000 -> h00_3006_FFFF] Width:unlimited Description:USB/SDMMC [RW]
[h00_3007_0000 -> h00_30FF_FFFF] Width:unlimited Description:Reserved [RW]
[h00_3100_0000 -> h00_3111_FFFF] Width:unlimited Description:MMIO [RW]
[h00_3112_0000 -> h00_37FF_FFFF] Width:unlimited Description:Reserved [RW]
[h00_3800_0000 -> h00_3800_FFFF] Width:unlimited Description:CLINT [RW]
[h00_3801_0000 -> h00_3801_FFFF] Width:unlimited Description:BEU [RW]
[h00_3802_0000 -> h00_3BFF_FFFF] Width:unlimited Description:Reserved []
[h00_3C00_0000 -> h00_3FFF_FFFF] Width:unlimited Description:PLIC [RW]
[h00_4000_0000 -> h00_7FFF_FFFF] Width:unlimited Description:PCIe [RW]
[h00_8000_0000 -> h1F_FFFF_FFFF] Width:unlimited Description:DDR [RWXIDSA]

Ftq: size:48
Roq: size:192 wbports:18 commitwidth:6
Regfile: size:160 read: 14write: 8
CSR: hasEmuPerfCnt:false

int wb arbiter:
[ AluExeUnit ] -> out #0
[ AluExeUnit ] -> out #1
[ AluExeUnit ] -> out #2
[ AluExeUnit ] -> out #3
[ LoadExu ] -> out #4
[ LoadExu ] -> out #5
[ MulDivExeUnit JmpExeUnit ] -> arb -> out #6
[ MulDivExeUnit FmiscExeUnit FmiscExeUnit ] -> arb -> out #7

Regfile: size:160 read: 14write: 8

fp wb arbiter:
[ FmacExeUnit ] -> out #0
[ FmacExeUnit ] -> out #1
[ FmacExeUnit ] -> out #2
[ FmacExeUnit ] -> out #3
[ LoadExu ] -> out #4
[ LoadExu ] -> out #5
[ FmiscExeUnit JmpExeUnit ] -> arb -> out #6
[ FmiscExeUnit ] ->  out #7

0: exu:FmacExeUnit fastPortsCnt: 4 slowPorts: 4 delay:4
1: exu:FmacExeUnit fastPortsCnt: 4 slowPorts: 4 delay:4
2: exu:FmacExeUnit fastPortsCnt: 4 slowPorts: 4 delay:4
3: exu:FmacExeUnit fastPortsCnt: 4 slowPorts: 4 delay:4
4: exu:FmiscExeUnit fastPortsCnt: 4 slowPorts: 4 delay:-1
5: exu:FmiscExeUnit fastPortsCnt: 4 slowPorts: 4 delay:-1
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at scala.collection.IndexedSeqOptimized.zipWithIndex(IndexedSeqOptimized.scala:106)
	at scala.collection.IndexedSeqOptimized.zipWithIndex$(IndexedSeqOptimized.scala:100)
	at scala.collection.immutable.StringOps.zipWithIndex(StringOps.scala:33)
	at chisel3.Printable$.pack(Printable.scala:84)
	at chisel3.printf$.printfWithoutReset(Printf.scala:101)
	at chisel3.assert$.$anonfun$apply_impl_do$2(Assert.scala:62)
	at chisel3.assert$$$Lambda$933/0x0000000840666040.apply(Unknown Source)
	at chisel3.WhenContext.<init>(When.scala:124)
	at chisel3.when$.apply(When.scala:31)
	at chisel3.assert$.apply_impl_do(Assert.scala:56)
	at freechips.rocketchip.tilelink.TLMonitor.monAssert(Monitor.scala:42)
	at freechips.rocketchip.tilelink.TLMonitor.$anonfun$legalizeFormatA$2(Monitor.scala:83)
	at freechips.rocketchip.tilelink.TLMonitor$$Lambda$979/0x0000000840688440.apply$mcV$sp(Unknown Source)
	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23)
	at chisel3.WhenContext.<init>(When.scala:124)
	at chisel3.when$.apply(When.scala:31)
	at freechips.rocketchip.tilelink.TLMonitor.legalizeFormatA(Monitor.scala:81)
	at freechips.rocketchip.tilelink.TLMonitor.$anonfun$legalizeFormat$2(Monitor.scala:369)
	at freechips.rocketchip.tilelink.TLMonitor$$Lambda$930/0x0000000840667c40.apply$mcV$sp(Unknown Source)
	at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:23)
	at chisel3.WhenContext.<init>(When.scala:124)
	at chisel3.when$.apply(When.scala:31)
	at freechips.rocketchip.tilelink.TLMonitor.legalizeFormat(Monitor.scala:369)
	at freechips.rocketchip.tilelink.TLMonitor.legalize(Monitor.scala:875)
	at freechips.rocketchip.tilelink.TLMonitorBase.<init>(Monitor.scala:22)
	at freechips.rocketchip.tilelink.TLMonitor.<init>(Monitor.scala:33)
	at freechips.rocketchip.tilelink.TLMonitorBuilder$$anonfun$$lessinit$greater$1.apply(Nodes.scala:11)
	at freechips.rocketchip.tilelink.TLMonitorBuilder$$anonfun$$lessinit$greater$1.apply(Nodes.scala:11)
	at freechips.rocketchip.tilelink.TLImp$.$anonfun$monitor$1(Nodes.scala:24)
	at freechips.rocketchip.tilelink.TLImp$$$Lambda$926/0x0000000840660c40.apply(Unknown Source)
	at chisel3.Module$.do_apply(Module.scala:54)
	at freechips.rocketchip.tilelink.TLImp$.monitor(Nodes.scala:24)
1 targets failed
XiangShan.test.runMain subprocess failed
make: *** [Makefile:49：build/XSTop.v] 错误 1

vitaminabcd, 2021-06-25:
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space

vitaminabcd, 2021-06-25:
我也遇到了，应该是这个原因：
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
我正在尝试使用更大内存容量的电脑。

QingYuDanYan, 2021-06-25:
试试运行以下命令后再 make一次。
export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xms256m -Xmx10g -Xss256K'

QingYuDanYan, 2021-06-25:
> 我也遇到了，应该是这个原因：
> Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
> 我正在尝试使用更大内存容量的电脑。

试试运行下面这条命令后再make。
export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xms256m -Xmx10g -Xss256K'

hyf6661669, 2021-06-25:
一般chisel源码越复杂，生成verilog需要的内存就越大，电脑内存不足就会报错，我记得之前看过生成rocket-chip和生成boom对应的verilog代码需要的内存就不同，boom需要的内存大很多，所以貌似只能换个大内存的电脑。。。

vitaminabcd, 2021-06-25:
> > 我也遇到了，应该是这个原因：
> > Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
> > 我正在尝试使用更大内存容量的电脑。
> 
> 试试运行下面这条命令后再make。
> export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xms256m -Xmx10g -Xss256K'

感谢建议。不过仍然报下面的异常：
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at java.base/java.lang.StringLatin1.newString(StringLatin1.java:715)

vitaminabcd, 2021-06-25:
有没有make verilog成功的伙伴？
你们的内存是多大？

hyf6661669, 2021-06-25:
开发人员能否把你们的服务器配置说一下作为参考呢？

vimcdoe, 2021-06-25:
> 开发人员能否把你们的服务器配置说一下作为参考呢？

https://github.com/OpenXiangShan/XiangShan/issues/838#issue-929315891

vitaminabcd, 2021-06-25:
> 问题可能是网络环境导致submodule初始化失败导致的。
> 
> 可以先尝试以下几个解决方案：
> (1) 再次make init
> (2) rm -rf out之后，重新make init
> (3) 删除rocket-chip, berkeley-hardfloat等submodule目录，重新make init
> 
> 如果仍然存在问题，可以试一下重新git clone XiangShan

是的，感谢，重新make init后解决了。

ljwljwljwljw, 2021-06-25:
JVM heap size不足导致的编译失败是由于之前的build.sc没能正确的将JVM Xmx参数传递下去， #842 可以修复该问题，经过测试10G heap足够编译香山，如果大家仍有问题请继续在issue中反馈，谢谢！

pxs7, 2021-06-25:
我按照 842 该了 build.sc 但是 编译 还是报相同错误；我的虚拟机 是 11G，  刚刚好也不行么

ljwljwljwljw, 2021-06-25:
> 我按照 842 该了 build.sc 但是 编译 还是报相同错误；我的虚拟机 是 11G， 刚刚好也不行么

能否再尝试增大xmx，例如11G、12G？在我们的部分机器上测试，xmx设为10G依然不够

haoranx98, 2021-06-28:
Ubuntu 20.04， 32GB内存，编译了半个多点吧 发送自 Windows 10 版邮件应用 发件人: vitaminabcd发送时间: 2021年6月25日 12:58收件人: OpenXiangShan/XiangShan抄送: Subscribed主题: Re: [OpenXiangShan/XiangShan] make verilog error (#837) 有没有make verilog成功的伙伴？你们的内存是多大？

pxs7, 2021-06-28:
这个可不可以优化一下，让编译时间短点------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年6月28日(星期一) 下午4:15
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [OpenXiangShan/XiangShan] make verilog error (#837)

sequencer, 2021-06-28:
立即购买M1

pxs7, 2021-06-28:
> 
> 立即购买M1

M1 有笔记本了么，不知道呀

abignail, 2021-09-03:
单运行make，就需要消耗15.4G的内存空间，电脑内存32G的，是毫无压力的。

aren0924, 2022-02-07:
请问 一下 ， 进入 xianshan/ , make init 后 ， make verilog , 遇到如下错误
[96/213] rocketchip.resolvedIvyDeps | Downloading [6/6] artifacts (~0/0 bytes)
1 targets failed
rocketchip.resolvedIvyDeps 
Resolution failed for 1 modules:
--------------------------------------------
  org.scalamacros:paradise_2.12.13:2.1.1 
	not found: /home/aren/.ivy2/local/org.scalamacros/paradise_2.12.13/2.1.1/ivys/ivy.xml
	download error: Caught java.net.UnknownHostException (repo1.maven.org) while downloading https://repo1.maven.org/maven2/org/scalamacros/paradise_2.12.13/2.1.1/paradise_2.12.13-2.1.1.pom
	download error: Caught java.net.UnknownHostException (oss.sonatype.org) while downloading https://oss.sonatype.org/content/repositories/snapshots/org/scalamacros/paradise_2.12.13/2.1.1/paradise_2.12.13-2.1.1.pom
	download error: Caught java.net.UnknownHostException (oss.sonatype.org) while downloading https://oss.sonatype.org/content/repositories/releases/org/scalamacros/paradise_2.12.13/2.1.1/paradise_2.12.13-2.1.1.pom

这个该怎么解决呢 ？？

poemonsense, 2022-02-07:
可以检查一下您那边的网络

aren0924, 2022-02-08:
> 可以检查一下您那边的网络
修改了 /etc/hosts , 把 我的 hostname 加在了 IP 后面。就没有问题了。
In fact , Caught java.net.UnknownHostException  基本上都是这个原因。
谢谢 啦 ！ ：）

cheng-zhen, 2022-03-17:
请问在./setup.sh时这些warn会有影响吗
/home/xxx/project/XiangShan/xs-env/NEMU/scripts/config.mk:5: Warning: .config does not exists!
-bash: /home/xxx/project/XiangShan/xs-env/NEMU/scripts/config.mk:5:: No such file or directory
xxx@105:~$ /home/xxx/project/XiangShan/xs-env/NEMU/scripts/config.mk:6: To build the porject, first run 'make menuconfig'.
-bash: /home/xxx/project/XiangShan/xs-env/NEMU/scripts/config.mk:6:: No such file or directory
...
class utils.FlushableQueueIO (1 calls): Unable to automatically infer cloneType on class utils.FlushableQueueIO. cloneType is now implemented by the Chisel compiler plugin so please ensure you are using it in your build. If you cannot use the compiler plugin or you are using it and you still see this message, please file an issue and let us know. For those not using the plugin, here is the 'runtime reflection' cloneType error message: constructor has parameters (entries) that are not both immutable and accessible. Either make all parameters immutable and accessible (vals) so cloneType can be inferred, or define a custom cloneType method.
There were 1 deprecated function(s) used. These may stop compiling in a future release - you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[warn]   In the sbt interactive console, enter:
[warn]     set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[warn]   or, in your build.sbt, add the line:
[warn]     scalacOptions := Seq("-unchecked", "-deprecation")
...
warning: ISO C++11 does not allow conversion from string literal to 'char *' [-Wwritable-strings]
    dumpGoldenMem("Init", track_instr, ticks);
...

另外在make verilog阶段，也会出现大量的warn
...
compiling 20 Scala sources to /home/xxx/project/XiangShan/xs-env/XiangShan/out/foreign-modules/rocket-chip/hardfloat/hardfloat/compile/dest/classes ...
[warn] 535 feature warnings; re-run with -feature for details
[warn] one warning found
[info] done compiling
[79/213] config.compile
[info] compiling 1 Scala source to /home/xxx/project/XiangShan/xs-env/XiangShan/out/foreign-modules/rocket-chip/api-config-chipsalliance/build-rules/mill/config/compile/dest/classes ...
[info] done compiling
[103/213] rocketchip.compile
[info] compiling 376 Scala sources to /home/xxx/project/XiangShan/xs-env/XiangShan/out/rocketchip/compile/dest/classes ...
[warn] two deprecations
[warn] four deprecations (since Chisel 3.5)
[warn] 16 deprecations (since FIRRTL 1.4)
[warn] 22 deprecations in total; re-run with -deprecation for details
[warn] 2573 feature warnings; re-run with -feature for details
[warn] 5 warnings found
[info] done compiling
...

history buffer length 256
[warning] update logic of foldest history has two or more levels of xor gates! histlen:16, compLen:8, at bit 6
[warning] update logic of foldest history has two or more levels of xor gates! histlen:16, compLen:8, at bit 7
[warning] update logic of foldest history has two or more levels of xor gates! histlen:32, compLen:8, at bit 6
[warning] update logic of foldest history has two or more levels of xor gates! histlen:32, compLen:8, at bit 7
[warning] update logic of foldest history has two or more levels of xor gates! histlen:119, compLen:8, at bit 6
[warning] update logic of foldest history has two or more levels of xor gates! histlen:32, compLen:11, at bit 9
[warning] update logic of foldest history has two or more levels of xor gates! histlen:13, compLen:7, at bit 5
...
最后无法生成verilog

poemonsense, 2022-03-17:
没有影响

hongyikl, 2022-07-07:
> 问题可能是网络环境导致submodule初始化失败导致的。
> 
> 可以先尝试以下几个解决方案： (1) 再次make init (2) rm -rf out之后，重新make init (3) 删除rocket-chip, berkeley-hardfloat等submodule目录，重新make init
> 
> 如果仍然存在问题，可以试一下重新git clone XiangShan

[image]
我这边make verilog遇到这个问题，以上全部试过了还是不行，求助

wakafa1, 2022-07-11:
https://github.com/OpenXiangShan/XiangShan/pull/1622 修复了这一问题，可以 pull 一下最新的修改

zhonglinghao, 2022-10-21:
我的环境是Ubuntu20.04LTS，按照开发环境部署配置，init以后make erilog出现以下报错问题，麻烦各位大佬指点谢谢
There is insufficient memory for the Java Runtime Environment to continue.
Native memory allocation (mmap) failed to map 19428016128 bytes for committing reserved memory.
An error report file with more information is saved as:
/home/zlh/桌面/xs-env/XiangShan/hs_err_pid18523.log
1 targets failed
XiangShan.runMain subprocess failed
make: *** [Makefile:57：build/XSTop.v] 错误 1

AugustusWillisWang, 2022-10-22:
> There is insufficient memory for the Java Runtime Environment to continue.

[RTFM](https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/troubleshoot-zh/)

closed 2023-02-15 by AugustusWillisWang

Chiwawachiwawa, 2023-04-04:
您好，我在進行 make verilog 這一步時，發現內存不夠，因此按照上述步驟將內存降為10G，不過我在non-inclusive L3這一步發現機器沒有反應(沒有報錯也沒有反應)，我大約兩次都等了兩個小時，也發現內存始終都保持著高占用的情況，請問是我等地不夠久還是需要另作調整(?
謝謝

Fred-Victor-YANG, 2023-11-02:
> 我的环境是Ubuntu20.04LTS，按照开发环境部署配置，init以后make erilog出现以下报错问题，麻烦各位大佬指点谢谢 There is insufficient memory for the Java Runtime Environment to continue. Native memory allocation (mmap) failed to map [...]

遇到一模一样的问题，已经尝试过以下所有办法：

1. rm -rf output; make init 无效
2. git submodule; rm -rf (all submodule); make init 无效
3. 重新 git clone 无效
4. 重新 git clone --recursive 无效
5. make bsp; make idea; make verilog 无效
6. ......

配置如下：

- 电脑是dell inspiron 15 3511
- 处理器	11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz   2.42 GHz
- 机带 RAM	16.0 GB (15.7 GB 可用)
- WSL2 Ubuntu 20.04
- vdisk 208G
[image]

请问该问题是拉取 submodule 有误还是电脑性能不行运存不足还是wsl虚拟机分配存储空间不足？如何解决？
谢谢！

wakafa1, 2023-11-02:
请在运行时监测一下内存占用情况，以排除内存过小的问题

Chiwawachiwawa, 2023-11-02:
makefile 要修改記憶體上限

yaoyuexiaogege, 2024-10-17:
mkdir -p build/rtl
time -avp -0 ./build/time.log mill -i xiangshan.runMain top.TopMain
seq-mem --repl-seq-mem-file=XSTop.sv.conf"
--target-dir build/rtl --config DefaultConfig --issue B--firtool-opt "--repl
--num-cores 1 --dump-fir --target systemverilog--split-verilog - firtool-opt
"-O=release - - disable-annotation-unknown - lowering-options=explicitBitcast,disallowLo
calVariables,disallowPortDeclSharing,locationInfoStyle=none" --fpga-platform --disable
-all --remove-assert --reset-gen --disable-always-basic-diff
/scripts/gen_sep_mem.sh "./scripts/vlsi_mem_gen" "buuild/rtl/XSTop.sv.conf" build/rtl
cat: build/rtl/XSTop.sv.conf: No such file or directory
rm: cannot remove 'build/rtl/XSTop.sv.conf.tmp': No such file or directory
make: *** [Makefile:155: build/rtl/XSTop.sv] Erгог 1
我这是什么原因？

eastonman, 2024-10-17:
> mkdir -p build/rtl
> time -avp -0 ./build/time.log mill -i xiangshan.runMain top.TopMain
> seq-mem --repl-seq-mem-file=XSTop.sv.conf"
> --target-dir build/rtl --config DefaultConfig --issue B--firtool-opt [...]

？？？

yaoyuexiaogege, 2024-10-17:
> ？？？

运行make verilog后出现的，请问我这个报错是什么原因？

zhang1career, 2025-01-09:
有没有小伙伴想在云服务上托管编译的？