OpenXiangShan/XiangShan #838: make verilog 期间内存溢出导致无法生成代码

Donlon, 2021-06-24:
Hi, 我在工程的根目录执行 `make verilog` 命令生成 Verilog 代码时，出现了 `java.lang.OutOfMemoryError: Java heap space` 错误，看起来是内存不够用了。不过运行时使用的电脑内存有 32G，按说是够用的，而且在出现错误前从资源监视器里也能看出内存并未用光（大概只占用了70%）。在 [build.sc#L84](https://github.com/OpenXiangShan/XiangShan/blob/master/build.sc#L84) 里有个 JVM 最大内存的选项，将它修改为 `-Xmx20G` 后问题也还是一样。不是很清楚 Scala/mill/Chisel 运行/编译时应该怎样配置内存相关的参数，在其他地方也没找到能解决问题的方法，希望能在这里得到一些帮助~

系统是 Manjaro 21.0.3，出错时显示的调用栈如下：
```
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at scala.collection.mutable.HashTable.resize(HashTable.scala:258)
	at scala.collection.mutable.HashTable.addEntry0(HashTable.scala:158)
	at scala.collection.mutable.HashTable.findOrAddEntry(HashTable.scala:170)
	at scala.collection.mutable.HashTable.findOrAddEntry$(HashTable.scala:167)
	at scala.collection.mutable.LinkedHashSet.findOrAddEntry(LinkedHashSet.scala:44)
	at scala.collection.mutable.LinkedHashSet.add(LinkedHashSet.scala:68)
	at scala.collection.mutable.LinkedHashSet.$plus$eq(LinkedHashSet.scala:63)
	at scala.collection.mutable.LinkedHashSet.$plus$eq(LinkedHashSet.scala:44)
	at scala.collection.generic.Growable.$anonfun$$plus$plus$eq$1(Growable.scala:62)
	at scala.collection.generic.Growable$$Lambda$12/1730173572.apply(Unknown Source)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
	at scala.collection.generic.Growable.$plus$plus$eq(Growable.scala:62)
	at scala.collection.generic.Growable.$plus$plus$eq$(Growable.scala:53)
	at scala.collection.mutable.AbstractSet.$plus$plus$eq(Set.scala:48)
	at scala.collection.mutable.SetLike.$plus$plus(SetLike.scala:191)
	at scala.collection.mutable.SetLike.$plus$plus$(SetLike.scala:191)
	at scala.collection.mutable.AbstractSet.$plus$plus(Set.scala:48)
	at firrtl.graph.DiGraph.getVertices(DiGraph.scala:57)
	at firrtl.transforms.DeadCodeElimination.run(DeadCodeElimination.scala:347)
	at firrtl.transforms.DeadCodeElimination.execute(DeadCodeElimination.scala:386)
	at firrtl.Transform.transform(Compiler.scala:319)
	at firrtl.Transform.transform$(Compiler.scala:319)
	at firrtl.transforms.DeadCodeElimination.transform(DeadCodeElimination.scala:33)
	at firrtl.stage.transforms.ExpandPrepares.execute(ExpandPrepares.scala:19)
	at firrtl.Transform.transform(Compiler.scala:319)
	at firrtl.Transform.transform$(Compiler.scala:319)
	at firrtl.stage.transforms.ExpandPrepares.transform(ExpandPrepares.scala:7)
	at firrtl.stage.transforms.CatchCustomTransformExceptions.execute(CatchCustomTransformExceptions.scala:10)
	at firrtl.Transform.transform(Compiler.scala:319)
	at firrtl.Transform.transform$(Compiler.scala:319)
1 targets failed
XiangShan.test.runMain subprocess failed
make: *** [Makefile:49: build/XSTop.v] Error 1
```

另外我在 GitHub Actions 上也试着跑了下，也会出现类似的错误，不过栈信息不太一样，可以在这里看到[完整的日志](https://github.com/Donlon/XiangShan/runs/2901213874)。

poemonsense, 2021-06-24:
我们之前在ubuntu 18.04的环境里面也遇到过这个问题，可以试一下配这个环境变量

`export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xms256m -Xmx10g -Xss256K'`

Donlon, 2021-06-24:
> 我们之前在ubuntu 18.04的环境里面也遇到过这个问题，可以试一下配这个环境变量
> 
> `export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xms256m -Xmx10g -Xss256K'`

加上这个配置后可以完整编译了，谢谢！

closed 2021-06-24 by Donlon

pxs7, 2021-06-25:
我用虚拟机编译也有这个问题， 编译至少虚拟机需要多少内存呢？

pxs7, 2021-06-26:
[deprecated] ListBuffer.scala:153 (5 calls): do_toBools is deprecated: "Use asBools instead"
[deprecated] ListBuffer.scala:164 (5 calls): do_toBools is deprecated: "Use asBools instead"
[warn] There were 2 deprecated function(s) used. These may stop compiling in a future release - you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[warn]   In the sbt interactive console, enter:
[warn]     set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[warn]   or, in your build.sbt, add the line:
[warn]     scalacOptions := Seq("-unchecked", "-deprecation")
Done elaborating.

这个界面停留了很久，build/XStop.v 这个文件也没有生成出来，这个是什么情况呢

Donlon, 2021-06-26:
> [deprecated] ListBuffer.scala:153 (5 calls): do_toBools is deprecated: "Use asBools instead"
> [deprecated] ListBuffer.scala:164 (5 calls): do_toBools is deprecated: "Use asBools instead"
> [warn] There [...]

Done elaborating 之后就会生成 Verilog 了，确实需要很久（我的电脑上就要差不多半小时），另外如果电脑内存不够的话可能会时不时卡在 GC 上导致特别慢，可以试试多等一会或者把内存调大一点。最小内存的话没试过，但最好还是给个 5-6 GB 以上的吧。