OpenXiangShan/XiangShan #839: 我是Java程序员 如何做贡献

xrfinbupt, 2021-06-25:
谢谢

acclea, 2021-06-25:
> 希望对RISC-V在中国的生态发展做一点点贡献
> 
> 要不要建一个微信群 或者 bbs

个人觉得你可以在这里出一个 RISC-5 入门指南【包括开发平台、开发工具、参考书籍等】，让有想法的潜在人员可以快速入门，可以进行简单的 demo 开发验证，然后逐渐的培养更多的潜在开发人员学习投入

pxs7, 2021-06-25:
能不能拉我进群

poemonsense, 2021-06-26:
稍等几天我们会开一个单独的文档仓库

ElegantLee, 2021-06-26:
> > 希望对RISC-V在中国的生态发展做一点点贡献
> > 要不要建一个微信群 或者 bbs
> 
> 个人觉得你可以在这里出一个 RISC-5 入门指南【包括开发平台、开发工具、参考书籍等】，让有想法的潜在人员可以快速入门，可以进行简单的 demo 开发验证，然后逐渐的培养更多的潜在开发人员学习投入

老哥的建议非常好，希望更多地国内开发者加入进来，为国内的技术生态做贡献！:muscle::muscle::muscle:

TurnOffNOD, 2021-06-27:
移植Java虚拟机、openjdk这些。

qinlili23333, 2021-06-28:
[阿里的riscv移植版android](https://github.com/T-head-Semi/aosp-riscv)目前就做到了10,或许可以考虑下把最新的12beta给port过来?

xrfinbupt, 2021-07-01:
> > 希望对RISC-V在中国的生态发展做一点点贡献
> > 要不要建一个微信群 或者 bbs
> 
> 个人觉得你可以在这里出一个 RISC-5 入门指南【包括开发平台、开发工具、参考书籍等】，让有想法的潜在人员可以快速入门，可以进行简单的 demo 开发验证，然后逐渐的培养更多的潜在开发人员学习投入

包云岗老师团队 翻译的《RISC-V 手册》我觉得就挺好的

closed 2023-10-21 by cebarobot