OpenXiangShan/XiangShan #847: 编译时间太久

pxs7, 2021-06-26:
[deprecated] ListBuffer.scala:153 (5 calls): do_toBools is deprecated: "Use asBools instead"
[deprecated] ListBuffer.scala:164 (5 calls): do_toBools is deprecated: "Use asBools instead"
[warn] There were 2 deprecated function(s) used. These may stop compiling in a future release - you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[warn] In the sbt interactive console, enter:
[warn] set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[warn] or, in your build.sbt, add the line:
[warn] scalacOptions := Seq("-unchecked", "-deprecation")
Done elaborating.
这个界面停留了很久，近1个小时， build/XStop.v 这个文件也没有生成出来，这个是什么情况呢

poemonsense, 2021-06-26:
一般5-20分钟能生成出来，如果时间过长，可能是内存不够导致发生了太多的交换，从而影响了生成速度。
可以试一下生成小版本的香山，`make verilog CONFIG=MinimalConfig`

pxs7, 2021-06-26:
编译过了

closed 2021-06-26 by pxs7