OpenXiangShan/XiangShan #848: 初学者 关于build.sc 疑问交流

pxs7, 2021-06-26:
object XiangShan extends CommonModule with SbtModule {
  override def millSourcePath = millOuterCtx.millSourcePath

  override def forkArgs = Seq("-Xmx10G")

  override def ivyDeps = super.ivyDeps() ++ chisel
  override def moduleDeps = super.moduleDeps ++ Seq(
    `rocket-chip`,
    `block-inclusivecache-sifive`,
    chiseltest
  )

  object test extends Tests {
    override def forkArgs = Seq("-Xmx10G")
    override def ivyDeps = super.ivyDeps() ++ Agg(
      ivy"org.scalatest::scalatest:3.2.0"
    )

    def testFrameworks = Seq(
      "org.scalatest.tools.Framework"
    )

    def testOnly(args: String*) = T.command {
      super.runMain("org.scalatest.tools.Runner", args: _*)
    }
  }

}

在 build.sc 里面xiangshan 依赖 是上面这么写，然后makefile里面 调用的是   XiangShan.test.runMain， 这个 是个 Xiangshan 类的类成员 test的 函数，然而 依赖放在 xiangshan这个类里面的， 我想请教一下  类的 类成员是如何从类得到依赖的，  还是 利用了scala 有什么特殊特性

ljwljwljwljw, 2021-06-26:
> object XiangShan extends CommonModule with SbtModule {
> override def millSourcePath = millOuterCtx.millSourcePath
>
> override def forkArgs = Seq("-Xmx10G")
>
> override def ivyDeps = super.ivyDeps() ++ [...]

build.sc并不是scala源文件，而是mill build配置文件，因此要按照 [mill](https://com-lihaoyi.github.io/mill/mill/Intro_to_Mill.html) 而不是scala的规则去使用，我们对test部分的编写参考自mill官方文档 [Addding a Test Suite](https://com-lihaoyi.github.io/mill/mill/Configuring_Mill.html#_adding_a_test_suite) 部分，mill相关问题请查阅mill文档。

sequencer, 2021-06-27:
TL;DR
> 我想请教一下 类的 类成员是如何从类得到依赖的， 还是 利用了scala 有什么特殊特性

I believe this piece code can solve your question:
https://github.com/com-lihaoyi/mill/blob/655ccdc04a10a120deb50fe1b643ecbec8a661bb/scalalib/src/ScalaModule.scala#L19-L25

This is a question related to build system:
Xiangshan use 2 build systems:
1. GNU Make
2. mill

There are 2 important part of a build system:
1. Frontend:
Actually build system is a DAG to describe a series of atomic tasks which consumes some inputs and generate a reproducible outputs.
Frontend defines its language describe this DAG and each task.
For more informations, [GNU Make](https://www.gnu.org/software/make/manual/html_node/Rules.html#Rules), [mill](https://com-lihaoyi.github.io/mill/mill/Tasks.html)
2. Backend:
Backend will execute those atomic actions, it will detect input changes, if some inputs are modified, task should be evaluated. if not changed, it will use caches last time, all mill build cache lays in `./out` folder.

So let's go back to your question:
> 在 build.sc 里面xiangshan 依赖 是上面这么写，然后makefile里面 调用的是 XiangShan.test.runMain， 这个 是个 Xiangshan 类的类成员 test的 函数，然而 依赖放在 xiangshan这个类里面的， 我想请教一下 类的 类成员是如何从类得到依赖的， 还是 利用了scala 有什么特殊特性

Xiangshan has 2 types of dependences: maven(ivy) and source dependence:
```
override def moduleDeps = super.moduleDeps ++ Seq(
  rocket-chip,
  block-inclusivecache-sifive,
  chiseltest
)
```
These are source dependence. Generally using this methodology is because Xiangshan did some changes to those library or those library didn't publish itself to maven.

Xiangshan depend on chisel3 in ivy:
```
override def ivyDeps = super.ivyDeps() ++ chisel
```
which means during the runtime, chisel will be downloaded from correspond maven repositories.

In mill, ivy will be cached by coursier, those package caches will be stored in `~/.cache/coursier`, since it can be share among projects.

pxs7, 2021-06-27:
trait ScalaModule extends JavaModule { outer =>

  trait Tests extends super.Tests with ScalaModule {

    override def scalaOrganization = outer.scalaOrganization()

    def scalaVersion = outer.scalaVersion()

    override def scalacPluginIvyDeps = outer.scalacPluginIvyDeps

    override def scalacPluginClasspath = outer.scalacPluginClasspath

    override def scalacOptions = outer.scalacOptions

  }}
这段代码感觉能够解释我的疑问了， scala 这个语言特性 能够直接定义 类.成员成员类 组合特质。 和一般语言差别太大了，多谢解答

closed 2021-06-27 by pxs7