OpenXiangShan/XiangShan #852: make emu error

Kevo-Gabo, 2021-06-26:
Hi,
 
Thanks for this great open source CPU.

After I successfully `make verilog`. I tried `make emu`. But the error occurs, and the error info is as follows.
Thank your very much for any suggestion.

> Connection:[DISPLAY_LOG_ENABLE] type:[func] source location:[SimTop] sink location:[AXI4RAM_1]
Connection:[logTimestamp] type:[func] source location:[SimTop] sink location:[AXI4RAM_1]
Connection:[XSPERF_CLEAN] type:[func] source location:[SimTop] sink location:[L2Prefetcher]
Connection:[XSPERF_DUMP] type:[func] source location:[SimTop] sink location:[L2Prefetcher]
[deprecated] ListBuffer.scala:153 (5 calls): do_toBools is deprecated: "Use asBools instead"
[deprecated] ListBuffer.scala:164 (5 calls): do_toBools is deprecated: "Use asBools instead"
[warn] There were 2 deprecated function(s) used. These may stop compiling in a future release - you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[warn]   In the sbt interactive console, enter:
[warn]     set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[warn]   or, in your build.sbt, add the line:
[warn]     scalacOptions := Seq("-unchecked", "-deprecation")
Done elaborating.
./scripts/vlsi_mem_gen build/SimTop.v.conf --output_file build/SimTop.v.sram.v
sed -i -e 's/$fatal/xs_assert(`__LINE__)/g' ./build/SimTop.v
Sat, 26 Jun 2021 17:38:50 +0800
time -a -o ./build/time.log verilator --cc --exe --top-module SimTop +define+VERILATOR=1 +define+PRINTF_COND=1 +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_DELAY=0 -Wno-STMTDLY -Wno-WIDTH -I/home/username/riscv/open_xiangshan/XiangShan/build --x-assign unique -O3 -CFLAGS "-std=c++11 -static -Wall -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest -DVERILATOR -Wno-maybe-uninitialized" -LDFLAGS "-lpthread -lSDL2 -ldl -lz" --assert --stats-vars --output-split 30000 --output-split-cfuncs 30000 \
        -o /home/username/riscv/open_xiangshan/XiangShan/build/emu -Mdir build/emu-compile build/SimTop.v ./src/test/vsrc/common/assert.v ./src/test/vsrc/common/difftest.v ./src/test/vsrc/common/ref.v ./src/test/vsrc/common/ram.v /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/emu.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/main.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/snapshot.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/compress.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/uart.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/ram.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/common.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/axi4.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/sdcard.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/vga.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/flash.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/keyboard.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/device.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/difftest.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/goldenmem.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/ref.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/nemuproxy.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/interface.cpp
Killed
%Error: Command Failed /usr/bin/verilator_bin --cc --exe --top-module SimTop \+define\+VERILATOR\=1 \+define\+PRINTF_COND\=1 \+define\+RANDOMIZE_REG_INIT \+define\+RANDOMIZE_MEM_INIT \+define\+RANDOMIZE_GARBAGE_ASSIGN \+define\+RANDOMIZE_DELAY\=0 -Wno-STMTDLY -Wno-WIDTH -I/home/username/riscv/open_xiangshan/XiangShan/build --x-assign unique -O3 -CFLAGS -std\=c\+\+11\ -static\ -Wall\ -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator\ -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common\ -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest\ -DVERILATOR\ -Wno-maybe-uninitialized -LDFLAGS -lpthread\ -lSDL2\ -ldl\ -lz --assert --stats-vars --output-split 30000 --output-split-cfuncs 30000 -o /home/username/riscv/open_xiangshan/XiangShan/build/emu -Mdir build/emu-compile build/SimTop.v ./src/test/vsrc/common/assert.v ./src/test/vsrc/common/difftest.v ./src/test/vsrc/common/ref.v ./src/test/vsrc/common/ram.v /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/emu.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/main.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/snapshot.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/compress.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/uart.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/ram.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/common.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/axi4.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/sdcard.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/vga.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/flash.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/keyboard.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/device.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/difftest.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/goldenmem.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/ref.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/nemuproxy.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/interface.cpp
make: *** [verilator.mk:107: build/emu-compile/VSimTop.mk] Error 137

poemonsense, 2021-06-26:
Is there any more error information when it crashes?

Did you ctrl-c to kill the process? I see from the log 
```
Killed
%Error: Command Failed
```

Kevo-Gabo, 2021-06-26:
> Is there any more error information when it crashes?
> 
> Did you ctrl-c to kill the process? I see from the log
> 
> ```
> Killed
> %Error: Command Failed
> ```

poemonsense, 2021-06-27:
Can you try a newer version of Verilator? Refer to Section `Git Quick Install` on https://verilator.org/guide/latest/install.html. 

v4.028 was released more than one year ago. We are currently using v4.204.

Kevo-Gabo, 2021-06-27:
> Can you try a newer version of Verilator? Refer to Section `Git Quick Install` on https://verilator.org/guide/latest/install.html.
> 
> v4.028 was released more than one year ago. We are currently using v4.204.

Tried `Verilator 4.204 2021-06-12 rev v4.204`, same error info.

> username@100013000737:~/riscv/open_xiangshan/XiangShan$ make emu
time -a -o ./build/time.log verilator --cc --exe --top-module SimTop +define+VERILATOR=1 +define+PRINTF_COND=1 +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_DELAY=0 -Wno-STMTDLY -Wno-WIDTH -I/home/username/riscv/open_xiangshan/XiangShan/build --x-assign unique -O3 -CFLAGS "-std=c++11 -static -Wall -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest -DVERILATOR -Wno-maybe-uninitialized" -LDFLAGS "-lpthread -lSDL2 -ldl -lz" --assert --stats-vars --output-split 30000 --output-split-cfuncs 30000 \
        -o /home/username/riscv/open_xiangshan/XiangShan/build/emu -Mdir build/emu-compile build/SimTop.v ./src/test/vsrc/common/assert.v ./src/test/vsrc/common/difftest.v ./src/test/vsrc/common/ref.v ./src/test/vsrc/common/ram.v /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/emu.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/main.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/snapshot.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/compress.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/uart.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/ram.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/common.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/axi4.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/sdcard.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/vga.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/flash.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/keyboard.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/device.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/difftest.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/goldenmem.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/ref.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/nemuproxy.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/interface.cpp
Killed
%Error: Command Failed /usr/local/bin/verilator_bin --cc --exe --top-module SimTop +define+VERILATOR=1 +define+PRINTF_COND=1 +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_DELAY=0 -Wno-STMTDLY -Wno-WIDTH -I/home/username/riscv/open_xiangshan/XiangShan/build --x-assign unique -O3 -CFLAGS -std=c++11\ -static\ -Wall\ -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator\ -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common\ -I/home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest\ -DVERILATOR\ -Wno-maybe-uninitialized -LDFLAGS -lpthread\ -lSDL2\ -ldl\ -lz --assert --stats-vars --output-split 30000 --output-split-cfuncs 30000 -o /home/username/riscv/open_xiangshan/XiangShan/build/emu -Mdir build/emu-compile build/SimTop.v ./src/test/vsrc/common/assert.v ./src/test/vsrc/common/difftest.v ./src/test/vsrc/common/ref.v ./src/test/vsrc/common/ram.v /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/emu.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/main.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/verilator/snapshot.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/compress.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/uart.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/ram.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/common.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/axi4.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/sdcard.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/vga.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/flash.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/keyboard.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/common/device.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/difftest.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/goldenmem.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/ref.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/nemuproxy.cpp /home/username/riscv/open_xiangshan/XiangShan/src/test/csrc/difftest/interface.cpp
make: *** [verilator.mk:107: build/emu-compile/VSimTop.mk] Error 137

poemonsense, 2021-07-01:
What's the configurations of your machine? For example, cpu, memory, storage?

Kevo-Gabo, 2021-07-10:
It's a laptop, WSL on Windows 10, Ubuntu 20.04.2 LTS, Intel Core i7-8565U @ 1.8GHz, 16.0 GB memory , 512GB SSD.
`make verilog` succeed.
`make emu` failed, and error info is [here](https://github.com/OpenXiangShan/XiangShan/issues/852#issuecomment-869123596).

Kevo-Gabo, 2021-07-10:
Tried on another machine: A cloud server, CentOS 8.2, 16.0 GB memory, 50GB SSD.

`make verilog` succeed with `export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xms256m -Xmx11g -Xss256K'`.
`make emu` still error.

**My environment**
```
[user@VM-0-17-centos XiangShan]# verilator --version
Verilator 4.204 2021-06-12 rev v4.204

[user@VM-0-17-centos XiangShan]# java -version
openjdk version "1.8.0_292"
OpenJDK Runtime Environment (build 1.8.0_292-b10)
OpenJDK 64-Bit Server VM (build 25.292-b10, mixed mode)

[user@VM-0-17-centos XiangShan]# g++ --version
g++ (GCC) 8.4.1 20200928 (Red Hat 8.4.1-1)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

[user@VM-0-17-centos XiangShan]# mill --version
Mill Build Tool version 0.9.8
Java version: 1.8.0_292, vendor: Red Hat, Inc., runtime: /usr/lib/jvm/java-1.8.0-openjdk-1.8.0.292.b10-1.el8_4.x86_64/jre
Default locale: en_US, platform encoding: UTF-8
OS name: "Linux", version: 4.18.0-193.28.1.el8_2.x86_64, arch: amd64
```

**Error output of `make emu`.**

> [user@VM-0-17-centos XiangShan]# make emu
Sat, 10 Jul 2021 21:13:37 +0800
time -a -o ./build/time.log verilator --cc --exe --top-module SimTop +define+VERILATOR=1 +define+PRINTF_COND=1 +define+RANDOMIZE_REG_INIT +define+RAN
DOMIZE_MEM_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_DELAY=0 -Wno-STMTDLY -Wno-WIDTH -I/user/riscv/OpenXiangShan/XiangShan/build --x-as
sign unique -O3 -CFLAGS "-std=c++11 -static -Wall -I/user/riscv/OpenXiangShan/XiangShan/src/test/csrc/verilator -I/user/riscv/OpenXiangShan/XiangShan
/src/test/csrc/common -I/user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest -DVERILATOR -Wno-maybe-uninitialized -DNUM_CORES=1" -LDFLAGS "-lpt
hread -lSDL2 -ldl -lz" --assert --stats-vars --output-split 30000 --output-split-cfuncs 30000 \
        -o /user/riscv/OpenXiangShan/XiangShan/build/emu -Mdir build/emu-compile build/SimTop.v ./src/test/vsrc/common/ref.v ./src/test/vsrc/common/a
ssert.v ./src/test/vsrc/common/difftest.v ./src/test/vsrc/common/ram.v /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/verilator/emu.cpp /user/risc
v/OpenXiangShan/XiangShan/src/test/csrc/verilator/snapshot.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/verilator/main.cpp /user/riscv/OpenX
iangShan/XiangShan/src/test/csrc/common/sdcard.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/uart.cpp /user/riscv/OpenXiangShan/XiangS
han/src/test/csrc/common/axi4.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/vga.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/
common/keyboard.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/device.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/ram.
cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/common.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/flash.cpp /user/risc
v/OpenXiangShan/XiangShan/src/test/csrc/common/compress.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest/goldenmem.cpp /user/riscv/Open
XiangShan/XiangShan/src/test/csrc/difftest/interface.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest/nemuproxy.cpp /user/riscv/OpenXia
ngShan/XiangShan/src/test/csrc/difftest/ref.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest/difftest.cpp
%Error: Verilator threw signal 9. Suggest trying --debug --gdbbt
%Error: Command Failed /usr/local/bin/verilator_bin --cc --exe --top-module SimTop +define+VERILATOR=1 +define+PRINTF_COND=1 +define+RANDOMIZE_REG_IN
IT +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_DELAY=0 -Wno-STMTDLY -Wno-WIDTH -I/user/riscv/OpenXiangShan/XiangSha
n/build --x-assign unique -O3 -CFLAGS -std=c++11\ -static\ -Wall\ -I/user/riscv/OpenXiangShan/XiangShan/src/test/csrc/verilator\ -I/user/riscv/OpenXi
angShan/XiangShan/src/test/csrc/common\ -I/user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest\ -DVERILATOR\ -Wno-maybe-uninitialized\ -DNUM_CO
RES=1 -LDFLAGS -lpthread\ -lSDL2\ -ldl\ -lz --assert --stats-vars --output-split 30000 --output-split-cfuncs 30000 -o /user/riscv/OpenXiangShan/Xiang
Shan/build/emu -Mdir build/emu-compile build/SimTop.v ./src/test/vsrc/common/ref.v ./src/test/vsrc/common/assert.v ./src/test/vsrc/common/difftest.v 
./src/test/vsrc/common/ram.v /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/verilator/emu.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/ve
rilator/snapshot.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/verilator/main.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/sd
card.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/uart.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/axi4.cpp /user/ri
scv/OpenXiangShan/XiangShan/src/test/csrc/common/vga.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/keyboard.cpp /user/riscv/OpenXiangS
han/XiangShan/src/test/csrc/common/device.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/ram.cpp /user/riscv/OpenXiangShan/XiangShan/sr
c/test/csrc/common/common.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/common/flash.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/co
mmon/compress.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest/goldenmem.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest
/interface.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest/nemuproxy.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest/re
f.cpp /user/riscv/OpenXiangShan/XiangShan/src/test/csrc/difftest/difftest.cpp
make: *** [verilator.mk:107: build/emu-compile/VSimTop.mk] Error 255

JustLuoyu, 2021-07-10:
I have a similar error :
$make emu CONFIG=MinimalConfig EMU_THREADS=2 -j16
xxx
%Error: Invalid Option: --threads-dpi

JustLuoyu, 2021-07-10:
> I have a similar error :
> $make emu CONFIG=MinimalConfig EMU_THREADS=2 -j16
> xxx
> %Error: Invalid Option: --threads-dpi
the version of Verilator which apt-get installed is too low. Git Quick Install solved my problem

poemonsense, 2021-07-11:
> Tried on another machine: A cloud server, CentOS 8.2, 16.0 GB memory, 50GB SSD.
>
> `make verilog` succeed with `export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xms256m -Xmx11g -Xss256K'`.
> `make emu` [...]

It looks like 16GB memory is not enough for verilator. I just tried on our machine (i9-10850k with 128GB memory). Verilator uses 31GB memory when I pass `--threads 8 --trace` to it. 

Sorry we didn't expect verilator requires so much memory. Can you try a larger swap size? Or using a machine with a larger memory?

Kevo-Gabo, 2021-07-12:
Finally `make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_THREADS=2 -j1` succeed, but it took me 16 hours with the swap size set to 32GB after 8/16GB failed.
(Change swap size by referring to https://net2.com/how-to-increase-swap-space-in-linux/).

By the way, I also install SDL2 to `make emu`, just paste the installing script below in case anyone need this.

> #!/bin/bash
if [[ $EUID -ne 0 ]]; then
   echo "This script must be run as root" 
   exit 1
fi
cd /tmp
git clone https://github.com/libsdl-org/SDL.git;
cd SDL;
mkdir build;
cd build;
../configure
make;
make install;
sudo ln -s /usr/local/lib/libSDL2-2.0.so.0 /usr/lib64/libSDL2-2.0.so.0

Kevo-Gabo, 2021-07-12:
So, can I draw a conclusion that if I don't have a machine with great performance(maybe 64GB, i9-10850K), I can hardly explore XiangShan and make contribution?

poemonsense, 2021-07-13:
> So, can I draw a conclusion that if I don't have a machine with great performance(maybe 64GB, i9-10850K), I can hardly explore XiangShan and make contribution?

Thanks and you have made contributions to XiangShan by filing an issue to us, which we didn't expect to be a real issue.

We have been working on this problem. Currently the diffculty for reproducing XiangShan on personal laptops is not what we want and what we expected. 

We have considered the following possibilities to allow users with their laptops to participate in XiangShan ecosystem:
- to use `CONFIG=MinimalSimConfig`, which replaces L1plus, PTW, DCache and lower level caches with DPI-C functions that does not produce RTL. This is acceptable for users who want to contribute to the OoO pipelines. 
- (WIP) to use the simple in-order NutShell (https://github.com/OSCPU/NutShell) as the cpu core and XiangShan's simulation framework (with difftest, a instruction-level co-sim and verification framework). We have started this project. This work will be done in one or two weeks (should be no later than end of July). For users with interest in our verification framework, this is OK.
- (WIP) to allow more configurations for XiangShan. `MinimalSimConfig` is probably not small enough but we have been working on it. We will let more pipelines to be configurable (especially the issue and function units) to achieve a two-issue or single-issue design. This work is expected to be done by the end of this month. 
- (WIP) use more DPI-C functions to replace the modules that users do not care. For example, if you want to see how the cache works, we let the CPU core be a functional model and only the caches are simulated based on RTL. This project is expected to be a long-term one (hope we can have it done by the end of this year).

If you have some ideas on how we can make this better, feel free to let us know.

BTW, there's no need to manually install the SDL lib (this lib is used in only one file and we rarely use it. Feel free to help us disable this file via some macro). Using `sudo apt install libsdl2-dev` to install by apt is more easier. 

Also, allowing `make` to use more threads via `-jN` like `-j4` helps the compilation a lot. Verilator produces hundreds of C++ files and we need to compile them all. If you have Synopsys VCS installed, it's much faster than Verilator since they work in different ways on simulation. It only takes less than 2 minutes to produce the `simv` though it simulates a lot slower than Verilator (hundreds of cycles per second in VCS v.s. 4K-6K cycles per second in Verilator with 8 threads).

poemonsense, 2021-07-17:
@Kevo-Gabo Hi, we have updated MinimalConfig on the master branch to include only 2 ALUs and 1FMAC. Could you please have a try to see how long it takes to generate verilog and the simulator? See `build/time.log` for the `time` result after it succeeds.

Kevo-Gabo, 2021-07-17:
Hi @poemonsense , thanks a lot for the update. Now, `make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_THREADS=2 -j8` takes about 15min.

closed 2021-07-17 by Kevo-Gabo

fatalfeel, 2021-08-19:
> > I have a similar error :
> > $make emu CONFIG=MinimalConfig EMU_THREADS=2 -j16
> > xxx
> > %Error: Invalid Option: --threads-dpi
> > the version of Verilator which apt-get installed is too low. Git Quick Install solved my problem

%Error: Invalid Option: --threads
apt remove verilator
git clone --recursive https://github.com/verilator/verilator
cd verilator
autoconf
./configure
make -j8 && make install

quanchenliu, 2024-11-19:
> Finally `make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_THREADS=2 -j1` succeed, but it took me 16 hours with the swap size set to 32GB after 8/16GB failed. (Change swap size by referring [...]

Thank you for your preparation work, which has saved us a lot of time.  
I would like to ask, when executing the command `make init`, is it necessary to extend the swap space to 32GB?

NewPaulWalker, 2024-11-20:
> I would like to ask, when executing the command `make init`, is it necessary to extend the swap space to 32GB?

Executing `make init` doesn't require that much space. This is because `init` is a phony target in the Makefile, designed to update and pull submodules. Generally, it only requires a stable network connection.

https://github.com/OpenXiangShan/XiangShan/blob/master/Makefile#L198