OpenXiangShan/XiangShan #855: Add this core to chipyard

HKalbasi, 2021-07-01:
[Chipyard](https://github.com/ucb-bar/chipyard) is a RISC-V SoC chisel framework. Since this core is also in chisel and open source, I think it is natural to add this core in chipyard. Is it possible to have it in chipyard? Will it be useful for chipyard and/or xiangshan and worth the needed efforts? I think it makes community of open source RISC-V cores more engaged in XiangShan.

ljwljwljwljw, 2021-07-02:
It is possible to add XiangShan into chipyard. For example, you can define a XiangShanTile by extending BaseTile, and then connect corresponding nodes between the tile and XiangShan core. But now the top-level connection part of Xiangshan's code is messy, we need to do a lot of code cleanup work first.

redpanda3, 2024-03-11:
https://github.com/redpanda3/xiangshan-chipyard @ljwljwljwljw @HKalbasi

redpanda3, 2024-03-18:
> https://github.com/redpanda3/xiangshan-chipyard @ljwljwljwljw @HKalbasi

Here is my solution