OpenXiangShan/XiangShan #864: make error : not found: value hardfloat

JustLuoyu, 2021-07-04:
$ make verilog
mkdir -p build
mill XiangShan.runMain top.TopMain -td build --config DefaultConfig --full-stacktrace --output-file XSTop.v --disable-all --remove-assert --infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf
Preparing Java 11.0.11 runtime; this may take a minute or two ...
Compiling /home/server/repo/XiangShan/build.sc
[27/191] api-config-chipsalliance.compile
Compiling compiler interface...
[info] compiling 1 Scala source to /home/server/repo/XiangShan/out/api-config-chipsalliance/compile/dest/classes ...
[info] done compiling
[51/191] rocket-chip.macros.compile
[info] compiling 1 Scala source to /home/server/repo/XiangShan/out/rocket-chip/macros/compile/dest/classes ...
[info] done compiling
[104/191] rocket-chip.compile
[info] compiling 376 Scala sources to /home/server/repo/XiangShan/out/rocket-chip/compile/dest/classes ...
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:293:25: not found: value hardfloat
[error]   def recode(x: UInt) = hardfloat.recFNFromFN(exp, sig, x)
[error]                         ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:294:23: not found: value hardfloat
[error]   def ieee(x: UInt) = hardfloat.fNFromRecFN(exp, sig, x)
[error]                       ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:465:25: not found: value hardfloat
[error]   val dcmp = Module(new hardfloat.CompareRecFN(maxExpWidth, maxSigWidth))
[error]                         ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:493:29: not found: value hardfloat
[error]       val conv = Module(new hardfloat.RecFNToIN(maxExpWidth, maxSigWidth, xLen))
[error]                             ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:503:35: not found: value hardfloat
[error]           val narrow = Module(new hardfloat.RecFNToIN(maxExpWidth, maxSigWidth, w))
[error]                                   ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:551:28: not found: value hardfloat
[error]       val i2f = Module(new hardfloat.INToRecFN(xLen, t.exp, t.sig))
[error]                            ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:560:55: value >> is not a member of Nothing
[error]     val dataPadded = data.init.map(d => Cat(data.last >> d.getWidth, d)) :+ data.last
[error]                                                       ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:560:60: value getWidth is not a member of Nothing
[error]     val dataPadded = data.init.map(d => Cat(data.last >> d.getWidth, d)) :+ data.last
[error]                                                            ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:614:35: not found: value hardfloat
[error]         val narrower = Module(new hardfloat.RecFNToRecFN(maxType.exp, maxType.sig, outType.exp, outType.sig))
[error]                                   ^
[error] /home/server/repo/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:648:46: not found: value hardfloat
[error]     val mulAddRecFNToRaw_preMul = Module(new hardfloat.MulAddRecFNToRaw_preMul(expWidth, sigWidth))
[error]                                              ^
[error] 14 errors found
1 targets failed
rocket-chip.compile Compilation failed
Makefile:48: recipe for target 'build/XSTop.v' failed
make: *** [build/XSTop.v] Error 1

poemonsense, 2021-07-05:
See #837

JustLuoyu, 2021-07-05:
solved! Berkeley-hardfloat is an empty folder.

closed 2021-07-05 by JustLuoyu