OpenXiangShan/XiangShan #869: 请教关于bpu中if3_prevHalfInstr的信号类型问题

daxxyy123, 2021-07-06:
我对于bpu中的if3_prevHalfInstr信号有个疑问。

在代码中if3_prevHalfInstr定义为reg，但是我觉得这个信号应该是wire？
if3_prevHalfInstr的赋值由从if4传过来的信号prevHalfInstrReq控制。（prevHalfInstrReq再控制信号if3_pendingPrevHalfInstr）
when (if3_prevHalfFlush) {
    if3_prevHalfInstr.valid := false.B
  }.elsewhen (hasPrevHalfInstrReq) {
    if3_prevHalfInstr.valid := true.B
  }.elsewhen (if3_prevHalfConsumed) {
    if3_prevHalfInstr.valid := false.B
  }
  when (hasPrevHalfInstrReq) {
    if3_prevHalfInstr.bits := prevHalfInstrReq.bits
  }
然后if3_prevHalfInstr再产生if3_prevHalfInstrMet信号。
val if3_prevHalfInstrMet = if3_pendingPrevHalfInstr && if3_prevHalfInstr.bits.npc === if3_pc && if3_valid
这里就产生问题了。
假设流水线指令如下
if1     if2    if3     if4
inst4 inst3 inst2 inst1
假设在if4的指令if4_pc(inst1)满足saveHalfRVI条件，则此时inst1相关的信号赋给prevHalfInstrReq，prevHalfInstrReq是wire，
prevHalfInstrReq.bits.npc := snpc(if4_pc)，如果预测正确的化，snpc(if4_pc)应该就是if3_pc（inst2）。
prevHalfInstrReq又被赋给if3_prevHalfInstr。
但是因为if3_prevHalfInstr是reg所以要延一个周期，一个周期后if3_pc已经变成inst3，而不是inst2了，真正的inst2已经跑到if4了。
这样 if3_prevHalfInstr.bits.npc === if3_pc这个等式其实就有问题了，此时if3_prevHalfInstr.bits.npc其实应该要等于if4_pc才是正确的。
但是如果if3_prevHalfInstr是wire，这个等式就是对的了。
请看一下，谢谢

Lingrui98, 2021-07-06:
这是一个好问题。

这里确实和我们的预期不符，可能会使跨取指packet的RVI指令引起额外的流水线冲刷。简单的改法是做一个bypass，但这样会在if4的预测路径后面加上额外的逻辑，可能导致时序问题。另外，即使做了bypass，一个相应的reg变量还是必要的，它可以处理流水线阻塞时的情况。

你可以尝试修复一下这个问题，欢迎提pull request :)

daxxyy123, 2021-07-08:
好的，明白了。

closed 2023-02-15 by AugustusWillisWang