OpenXiangShan/XiangShan #870: 下载XiangShan子模块无法下载berkeley-softfloat-3

zhuxuanlong, 2021-07-07:
您好！
我的环境是公司内网，用的是我司的服务器，CPU是双路 Intel(R) Xeon(R) Silver 4114 CPU @ 2.20GHz，内存376G。下载XiangShan代码，make init命令提示已经全部下载完毕，但通过git submodule update --init --recursive 等手动命令提示无法下载berkeley-softfloat-3。我在网页上通过登录XiangShan代码仓库是可以直接打开此代码的。
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# make init
git submodule update --init
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git submodule update --init --recursive
正克隆到 'berkeley-softfloat-3'...
fatal: unable to connect to github.com:
github.com[0: 52.74.223.119]: errno=连接超时

fatal: 无法克隆 'git://github.com/ucb-bar/berkeley-softfloat-3.git' 到子模组路径 'berkeley-softfloat-3'
无法递归进子模组路径 'berkeley-hardfloat'

root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git remote -v
origin  https://github.com/OpenXiangShan/XiangShan.git (fetch)
origin  https://github.com/OpenXiangShan/XiangShan.git (push)

poemonsense, 2021-07-07:
看着像是网络的问题，因为提示是连接超时。

其实只要make init就可以，不需要--recursive，建议参考 https://github.com/OpenXiangShan/XiangShan/issues/837#issuecomment-867667342

pascalApple, 2021-07-07:
换ssh协议，也就是git config --global url."git@github.com:".Insteadof https://github.com/

zhuxuanlong, 2021-07-07:
> 看着像是网络的问题，因为提示是连接超时。
> 
> 其实只要make init就可以，不需要--recursive，建议参考 [#837 (comment)](https://github.com/OpenXiangShan/XiangShan/issues/837#issuecomment-867667342)

如果只看make init，已经提示全部下载成功了吧。因为此时没有提示还有什么没有下载的。

poemonsense, 2021-07-07:
- git status
- cd berkeley-hardfloat && git status

可以看一下这两个状态

zhuxuanlong, 2021-07-07:
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git status
位于分支 master
您的分支与上游分支 'origin/master' 一致。
无文件要提交，干净的工作区
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# cd berkeley-hardfloat/
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan/berkeley-hardfloat#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan/berkeley-hardfloat# git status
头指针分离于 8031f85
无文件要提交，干净的工作区

我参照上面建议，参考#837，我删除了berkeley-hardfloat，也git清除了相关信息，make init，还是看不到berkeley-softfloat-3下面有任何内容：

root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# rm -rf berkeley-hardfloat/
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git clean -fd && git submodule foreach --recursive git clean -fd
正在进入 'api-config-chipsalliance'
正在进入 'block-inclusivecache-sifive'
正在进入 'chiseltest'
正在进入 'ready-to-run'
正在进入 'rocket-chip'
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# make init
git submodule update --init
子模组路径 'berkeley-hardfloat'：检出 '8031f8594278967d57016ba52a48393662abeb05'
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# ls berkeley-hardfloat/
berkeley-softfloat-3  berkeley-testfloat-3  build.sbt  build.sc  doc  LICENSE  Makefile  project  README.md  sbt-launch.jar  src
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# ls berkeley-hardfloat/berkeley-softfloat-3/
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git status
位于分支 master
您的分支与上游分支 'origin/master' 一致。
无文件要提交，干净的工作区

pascalApple, 2021-07-07:
> root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git status
> 位于分支 master
> 您的分支与上游分支 'origin/master' 一致。
> 无文件要提交，干净的工作区
> root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# [...]

传个ssh密钥上来用ssh协议就完事了，https会遇到些玄学的问题，我前几天就用ssh clone的

poemonsense, 2021-07-07:
> 我参照上面建议，参考#837，我删除了berkeley-hardfloat，也git清除了相关信息，make init，还是看不到berkeley-softfloat-3下面有任何内容：

git submodule init的时候不需要--recursive，所以不需要有softfloat。
https://github.com/OpenXiangShan/XiangShan/issues/870#issuecomment-875206588 提到了，只需要make init, which并没有--recursive

zhuxuanlong, 2021-07-07:
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git status
位于分支 master
您的分支与上游分支 'origin/master' 一致。
无文件要提交，干净的工作区
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# cd berkeley-hardfloat/
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan/berkeley-hardfloat#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan/berkeley-hardfloat# git status
头指针分离于 8031f85
无文件要提交，干净的工作区

我参照上面建议，参考#837，我删除了berkeley-hardfloat，也git清除了相关信息，make init，还是看不到berkeley-softfloat-3下面有任何内容：

root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# rm -rf berkeley-hardfloat/
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git clean -fd && git submodule foreach --recursive git clean -fd
正在进入 'api-config-chipsalliance'
正在进入 'block-inclusivecache-sifive'
正在进入 'chiseltest'
正在进入 'ready-to-run'
正在进入 'rocket-chip'
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# make init
git submodule update --init
子模组路径 'berkeley-hardfloat'：检出 '8031f8594278967d57016ba52a48393662abeb05'
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# ls berkeley-hardfloat/
berkeley-softfloat-3  berkeley-testfloat-3  build.sbt  build.sc  doc  LICENSE  Makefile  project  README.md  sbt-launch.jar  src
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# ls berkeley-hardfloat/berkeley-softfloat-3/
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/XiangShan# git status
位于分支 master
您的分支与上游分支 'origin/master' 一致。
无文件要提交，干净的工作区

> > 我参照上面建议，参考#837，我删除了berkeley-hardfloat，也git清除了相关信息，make init，还是看不到berkeley-softfloat-3下面有任何内容：
> 
> git submodule init的时候不需要--recursive，所以不需要有softfloat。
> [#870 (comment)](https://github.com/OpenXiangShan/XiangShan/issues/870#issuecomment-875206588) 提到了，只需要make init, which并没有--recursive

也就是说Master分支，没有软浮点是不影响正常编译仿真的，是吧？

poemonsense, 2021-07-07:
> 也就是说Master分支，没有软浮点是不影响正常编译仿真的，是吧？

可以试一下，这样就知道影不影响了

Piping, 2021-07-14:
试试 `git submodule update --init --recursive`

closed 2021-07-24 by zhuxuanlong

zhuxuanlong, 2021-07-24:
那么就出现另外一个问题，我昨天新下载代码，make init，并没有提示需要更新什么了，但执行make emu报错：
root@zte-R5300-G4-riscv:/share/zhuxuanlong/xiangshan_master/XiangShan# make init
git submodule update --init
root@zte-R5300-G4-riscv:/share/zhuxuanlong/xiangshan_master/XiangShan#

root@zte-R5300-G4-riscv:/share/zhuxuanlong/xiangshan_master/XiangShan# make emu SIM_ARGS=--disable-log EMU_THREADS=16
mkdir -p build
Sat, 24 Jul 2021 02:26:13 +0800
time -a -o ./build/time.log mill XiangShan.test.runMain top.SimTop -td build
--config DefaultConfig --full-stacktrace --output-file SimTop.v
--infer-rw --repl-seq-mem -c:top.SimTop:-o:build/SimTop.v.conf
--num-cores 1 --disable-log

Compiling /share/zhuxuanlong/xiangshan_master/XiangShan/build.sc
[27/213] api-config-chipsalliance.compile
Compiling compiler interface...
[info] compiling 1 Scala source to /share/zhuxuanlong/xiangshan_master/XiangShan/out/api-config-chipsalliance/compile/dest/classes ...
[info] done compiling
[51/213] rocket-chip.macros.compile
[info] compiling 1 Scala source to /share/zhuxuanlong/xiangshan_master/XiangShan/out/rocket-chip/macros/compile/dest/classes ...
[info] done compiling
[104/213] rocket-chip.compile
[info] compiling 376 Scala sources to /share/zhuxuanlong/xiangshan_master/XiangShan/out/rocket-chip/compile/dest/classes ...
[error] /share/zhuxuanlong/xiangshan_master/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:293:25: not found: value hardfloat
[error] def recode(x: UInt) = hardfloat.recFNFromFN(exp, sig, x)
[error] ^

我看代码，确实伯克利硬浮点没有下全。

也就是碰到这种问题，如何进行？

Piping, 2021-07-24:
git 没下全有提示的，需要检查终端输出。现在最好重新克隆把步骤做一遍