OpenXiangShan/XiangShan #873: 关于icache中predecode的问题

daxxyy123, 2021-07-08:
val pds = Seq.fill(nWays)(Module(new PreDecode))
  for (i <- 0 until nWays) {
    val wayResp = Wire(new ICacheResp)
    val wayData = cutHelper(s3_data(i), s3_req_pc, s3_req_mask)
    val refillData = Mux(s3_sec_miss,cutHelper(refillDataVecReg, s3_req_pc,s3_req_mask),cutHelper(refillDataVec, s3_req_pc,s3_req_mask))
    wayResp.pc := s3_req_pc
    wayResp.data := Mux(s3_valid && s3_hit, wayData, Mux(s3_mmio ,mmio_packet ,refillData))
    wayResp.mask := s3_req_mask
    wayResp.ipf := s3_exception_vec(pageFault)
    wayResp.acf := s3_exception_vec(accessFault)
    wayResp.mmio := s3_mmio
    pds(i).io.in := wayResp
    pds(i).io.prev <> io.prev
    pds(i).io.prev_pc := io.prev_pc
  }

  io.pd_out := Mux1H(s3_wayMask, pds.map(_.io.out))

icache的设计中针对每一个way都实例化了一个predecode，但是其实每次只会有1个hit way，按道理应该通过s3_wayMask筛选一下，然后只针对真正的hit way来做predecode。
不知道现在这个设计有什么特殊的考虑？谢谢

Lingrui98, 2021-07-08:
这样predecode和多路很长的数据的选取可以并行，对时序有好处

daxxyy123, 2021-07-09:
谢谢，还是有点不大明白，再多问一句：原先的时序路径是：4 way data ram  输出--各路中间处理--各路predecode--选择器；如果改成先选的化是：4 way data ram  输出 --选择器 -- 选中路中间处理--单独predecode。貌似最长的数据路径没啥变化？为啥并行处理数据路径会变短，最后还是要走mux绕一圈呀？

Lingrui98, 2021-07-09:
更改前后的选择器位宽是不同的，一种是对整个cache line的数据宽度进行四选一，延迟大，而另一种是对预译码的结果进行选择，位宽小延迟低

daxxyy123, 2021-07-09:
谢谢，明白了，整cacheline的宽度确实大太大了，直接做mux是不合理，那是否应该对于所有的way，做完中间处理，截取需要的packet，基于packet做mux，以后再做predecode呀？这样的mux是packet宽度，比现在基于pedecode输出的要小些而且predecode更少了？

Lingrui98, 2021-07-09:
packet宽度是cache line的一半，仍然很大，predecode出来的做选择的数据在chisel生成的verilog里最宽是39位，虚拟地址的宽度。
这里多例化预译码模块的目的本质上是面积换时序，到底如何选择可以根据具体设计做权衡

closed 2021-11-18 by daxxyy123