OpenXiangShan/XiangShan #876: 制作bbl.bin失败

zhuxuanlong, 2021-07-12:
我是按照Linux Kernel 的构建.md里面的方法想自己构建一个bbl.bin。
1、riscv-linux我是下的压缩包：riscv-linux-nanshan.zip。编译vmlinux，来看也编译出来了。
drwxr-sr-x   4 root nopasswdlogin    4096 7月  12 13:16 virt
-rwxr-xr-x   1 root nopasswdlogin 1401056 7月  12 13:16 vmlinux
-rw-r--r--   1 root nopasswdlogin     112 7月  12 13:16 .vmlinux.cmd
-rw-r--r--   1 root nopasswdlogin 4569448 7月  12 13:16 vmlinux.o

2、riscv-rootfs，里面的hello也编译了

3、riscv-pk这块编译错误：
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk/dts# rm platform.dtsi
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk/dts# ln -s noop.dtsi platform.dtsi
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk/dts# ls -al
总用量 28
drwxr-sr-x  2 root nopasswdlogin 4096 7月  12 13:20 .
drwxr-sr-x 11 root nopasswdlogin 4096 7月   1 15:09 ..
-rw-r--r--  1 root nopasswdlogin  801 7月   1 15:09 nemu.dtsi
-rw-r--r--  1 root nopasswdlogin  691 7月   1 15:09 noop.dtsi
lrwxrwxrwx  1 root nopasswdlogin    9 7月  12 13:20 platform.dtsi -> noop.dtsi
-rw-r--r--  1 root nopasswdlogin 4012 7月   1 15:09 PXIe.dtsi
-rw-r--r--  1 root nopasswdlogin 1249 7月   1 15:09 system.dts
-rw-r--r--  1 root nopasswdlogin 2201 7月   1 15:09 zynq-standalone.dtsi

root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk# make clean
rm -rf build
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk#
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk# make noop -j
RISCV_ROOTFS_HOME=/share/zhuxuanlong/XiangShan/riscv-rootfs make -C /share/zhuxuanlong/XiangShan/riscv-rootfs
mkdir -p build
mkdir -p build
dtc -O dtb -I dts -o build/system.dtb dts/system.dts
cd build && /share/zhuxuanlong/XiangShan/riscv-pk/configure --host=riscv64-unknown-elf --with-payload=/share/zhuxuanlong/XiangShan/riscv-linux-nanshan/riscv-linux-nanshan/vmlinux --with-arch=rv64imac --enable-logo
Error: dts/platform.dtsi:1.1-9 syntax error
FATAL ERROR: Unable to parse input tree
make[1]: 进入目录“/share/zhuxuanlong/XiangShan/riscv-rootfs”
make: *** [Makefile:60：build/system.dtb] 错误 1
make: *** 正在等待未完成的任务....
echo apps/hello
echo apps/stream
echo apps/busybox
echo apps/redis
apps/hello
echo apps/dwarf/md5
apps/stream
make -s -C apps/hello install
apps/busybox
make -s -C apps/stream install
apps/redis
make -s -C apps/busybox install
apps/dwarf/md5
make -s -C apps/redis install
make -s -C apps/dwarf/md5 install
echo apps/dwarf/sort
echo apps/dwarf/wordcount
make[2]: 进入目录“/share/zhuxuanlong/XiangShan/riscv-rootfs/apps/hello”
apps/dwarf/sort
make -s -C apps/dwarf/sort install
make[2]: 进入目录“/share/zhuxuanlong/XiangShan/riscv-rootfs/apps/stream”
apps/dwarf/wordcount

wakafa1, 2021-07-12:
出错信息显示是 dtc 在编译设备树的时候产生了问题，但我这边无法复现。请运行一下 `dtc -O dtb -I dts -o build/system.dtb dts/system.dts` 看看是否仍然有同样的问题。另外，麻烦提供一下 dtc 的版本

zhuxuanlong, 2021-07-13:
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk# dtc -v
Version: DTC 1.4.0

root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk# dtc -O dtb -I dts -o build/system.dtb dts/system.dts
Error: dts/platform.dtsi:1.1-9 syntax error
FATAL ERROR: Unable to parse input tree
root@zte-R5300-G4:/share/zhuxuanlong/XiangShan/riscv-pk#

zhuxuanlong, 2021-07-13:
root:/XiangShan/riscv-pk# cat dts/platform.dtsi
/dts-v1/;

/ {
        L18: soc {
                L2: clint@2000000 {
                        compatible = "riscv,clint0";
                        interrupts-extended = <&L4 3 &L4 7>;
                        reg = <0x0 0x38000000 0x0 0x10000>;
                        reg-names = "control";
                        clock-frequency-mhz = <500>;
                };
                L14: serial@60000000 {
                        compatible = "xilinx,uartlite";
                        //interrupt-parent = <&L1>;
                        interrupts = <3>;
                        reg = <0x0 0x40600000 0x0 0x1000>;
                        reg-names = "control";
                };
    sdhci: mmc {
      compatible = "nemu-sdhost";
                        reg = <0x0 0x40002000 0x0 0x1000>;
    };
        };
  chosen {
    bootargs = "root=/dev/mmcblk0 rootfstype=ext4 ro rootwait earlycon";
  };
        L11: memory@100000000 {
                device_type = "memory";
                reg = <0x0 0x80000000 0x0 0x2000000>;
        };
};

wakafa1, 2021-07-13:
怀疑是 dtc 的版本问题，请使用 1.5.0 版本的 dtc 试一下

zhuxuanlong, 2021-07-24:
dtc版本为1.5.0后就正常了。这块建议在代码中约束一下dtc版本依赖。

closed 2021-07-24 by wakafa1