OpenXiangShan/XiangShan #878: 编译verilog CONFIG=MinimalConfig 出错

QianpengLi577, 2021-07-13:
[deprecated] ListBuffer.scala:153 (4 calls): do_toBools is deprecated: "Use asBools instead"
[deprecated] ListBuffer.scala:164 (4 calls): do_toBools is deprecated: "Use asBools instead"
[warn] There were 2 deprecated function(s) used. These may stop compiling in a future release - you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[warn]   In the sbt interactive console, enter:
[warn]     set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[warn]   or, in your build.sbt, add the line:
[warn]     scalacOptions := Seq("-unchecked", "-deprecation")
Done elaborating.
xiangshan.backend.MemBlock@45e34a32
./scripts/vlsi_mem_gen build/XSTop.v.conf --tsmc28 --output_file build/tsmc28_sram.v > build/tsmc28_sram.v.conf
./scripts/vlsi_mem_gen build/XSTop.v.conf --output_file build/sim_sram.v
# sed -i -e 's/_\(aw\|ar\|w\|r\|b\)_\(\|bits_\)/_\1/g' build/XSTop.v
fatal: 您的当前分支 'master' 尚无任何提交
make: *** [Makefile:54：build/XSTop.v] 错误 128

请问这是什么情况呢？

poemonsense, 2021-07-13:
麻烦提供一下git status和git log -n 1的结果，以及您那边makefile 54行的内容。根据现在提供的出错信息，我们没办法复现，所以也给不出原因

QianpengLi577, 2021-07-14:
# git status
```
位于分支 master

尚无提交

未跟踪的文件:
  （使用 "git add <文件>..." 以包含要提交的内容）
	.github/
	.gitignore
	.gitmodules
	.mill-version
	LICENSE
	Makefile
	README.md
	api-config-chipsalliance/
	berkeley-hardfloat/
	block-inclusivecache-sifive/
	build.sbt
	build.sc
	chiseltest/
	debug/
	fpga/
	project/
	readme.zh-cn.md
	ready-to-run/
	rocket-chip/
	scalastyle-config.xml
	scalastyle-test-config.xml
	scripts/
	src/
	tools/
	vcs.mk
	verilator.mk
	xs-arch-simple.svg

提交为空，但是存在尚未跟踪的文件（使用 "git add" 建立跟踪）
```

# git log -n 1

`fatal: 您的当前分支 'master' 尚无任何提交`

# makefile 48~64行

```
$(TOP_V): $(SCALA_FILE)
        mkdir -p $(@D)
        mill XiangShan.runMain $(FPGATOP) -td $(@D)                      \
                --config $(CONFIG) --full-stacktrace --output-file $(@F)     \
                --disable-all --remove-assert --infer-rw                     \
                --repl-seq-mem -c:$(FPGATOP):-o:$(@D)/$(@F).conf $(SIM_ARGS) \
                --num-cores $(NUM_CORES)
        $(MEM_GEN) $(@D)/$(@F).conf --tsmc28 --output_file $(@D)/tsmc28_sram.v > $(@D)/tsmc28_sram.v.conf
        $(MEM_GEN) $(@D)/$(@F).conf --output_file $(@D)/sim_sram.v
        # sed -i -e 's/_\(aw\|ar\|w\|r\|b\)_\(\|bits_\)/_\1/g' $@
        @git log -n 1 >> .__head__
        @git diff >> .__diff__
        @sed -i 's/^/\/\// ' .__head__
        @sed -i 's/^/\/\//' .__diff__
        @cat .__head__ .__diff__ $@ > .__out__
        @mv .__out__ $@
        @rm .__head__ .__diff__
```

54行为：
`--num-cores $(NUM_CORES)`

poemonsense, 2021-07-14:
看起来这里是一个新仓库，而不是 `git clone` 下来的，导致 `git log -n 1` 这一行命令出错了。

我们强烈建议使用香山时采用 `git clone` 来下载，而不是下载压缩包。因为 submodule 的初始化需要有 git 环境。
或者您也可以把 Makefile 里面这两行删掉
```
@git log -n 1 >> .__head__
@git diff >> .__diff__
```

QianpengLi577, 2021-07-15:
您好，我重新`git clone`了整个仓库，运行`make verilog CONFIG=MinimalConfig`，出现了如下错误

```
Done elaborating.
xiangshan.backend.MemBlock@3d1b27ec
./scripts/vlsi_mem_gen build/XSTop.v.conf --tsmc28 --output_file build/tsmc28_sram.v > build/tsmc28_sram.v.conf
/usr/bin/env: python3: 没有那个文件或目录
make: *** [build/XSTop.v] 错误 127
```

但是build文件夹已经生成了如下文件

```
firrtl_black_box_resource_files.f  time.log            XSTop.0x38000000.0.regmap.json  XSTop.0x3c000000.0.regmap.json  XSTop.dts  XSTop.v
plusarg_reader.v                   tsmc28_sram.v.conf  XSTop.0x38010000.0.regmap.json  XSTop.anno.json                 XSTop.fir  XSTop.v.conf
```

请问我接下来该怎么做呢？

poemonsense, 2021-07-15:
> /usr/bin/env: python3: 没有那个文件或目录
错误日志显示，python3这个命令没有找到，可以安装一下 python3。

不过，Verilog其实已经生成了，在XSTop.v，只是其中的SRAM仿真文件(一部分模块)没有生成，`make verilog`在Chisel这一步已经完成了

QianpengLi577, 2021-07-15:
谢谢！

closed 2021-07-15 by QianpengLi577