OpenXiangShan/XiangShan #879: 关于freelist的问题

daxxyy123, 2021-07-14:
rename模块中的freelist是一个128项的环形队列，freelist中放置可使用的free reg，headPtr指向头，从headPtr开始分配，把free reg分配给指令；tailPtr指向尾，释放的reg从tailPtr重新加入freelist。正常情况两者同时往前走，当然tailPtr不能超越headPtr，但是因为存在mispredict的情况，headPtr某些情况下需要回滚。这样似乎会带来一个问题，假如headPtr和tailPtr靠的比较近，当headPtr需要回滚到位置x时，却发现x已经被tailPtr超越，即本来准备回滚重新分配的freelist值，已经被释放的reg值覆盖了，这些值将永远无法重新在freelist中出现。此时虽然core可以继续运作，但是实际上部分物理reg已经泄露，无法被再分配了。
不知道我理解的是否正确，这种场景是否会出现？
谢谢！

ljwljwljwljw, 2021-07-14:
tailPtr是在ROB中的指令提交时，将其旧的目的物理寄存器(old_pdest)释放，tailPtr+1，headPtr在mispredict时最终恢复的位置是误预测的那条指令之前的位置，一定晚于tailPtr对应的已经提交的指令

YikeZhou, 2021-07-19:
这种情况是不可能出现的。请看 free list 的源码：

```scala
val FL_SIZE = NRPhyRegs - 32
```

也就是说：free list 的容量 (128) + 逻辑寄存器数 (32) = 物理寄存器数 (160)

这是因为，任意时刻下，物理寄存器中都有 32 个寄存器存放着**逻辑寄存器**的值，这里的逻辑寄存器是指，按照正确程序顺序串行执行指令到最后一条 commit 的指令时，体系结构所规定的 32 个寄存器的正确值。也就是说，最多只能有 128 个空闲的物理寄存器，这也是能够使用 ring buffer 管理空闲寄存器的前提。

所谓的“回滚”是 cpu 发现 roq 中的一些指令不在正确的执行路径上，需要取消它们的执行。之前分配给它们存放结果的物理寄存器也可以释放。本来空闲的物理寄存器就最多有 128 个，即使 roq 中所有的指令都需要取消执行（flush），也只是把 head ptr 移到 tail ptr 处而已。

YikeZhou, 2021-07-19:
一条指令 rename 时从 free list 里得到一个物理寄存器编号作为 `pdst`，并记录 spec RAT 里的 `old_pdst`。

该指令 commit 时，前面的所有指令已经 commit 完毕，此时的 arch RAT 就是 rename 时的 spec RAT。这条指令提交的过程中

- `old_pdst` 被写入 free list 的 tail ptr 处
- `pdst` 写入 arch RAT 中，覆盖 `old_pdst`

这样就完成了 free list 和 arch RAT 中数据的交换。

无论何时，一个物理寄存器的编号，要么在 free list 中，要么在 arch RAT 中，不存在**物理 reg 泄露**的情况。

closed 2021-11-18 by daxxyy123