OpenXiangShan/XiangShan #881: mill installation method has expired

liliwenaaa, 2021-07-15:
I can't install mill in my ubuntu

poemonsense, 2021-07-15:
Refer to https://com-lihaoyi.github.io/mill/mill/Intro_to_Mill.html#_installation for installation of mill. 

If you cannot install mill with these instructions, please provide more information and we will try to reproduce your steps.

wakafa1, 2021-07-15:
It turns out that official link of mill is unavailable.
You may try the following command: 
`sudo sh -c "curl -L https://github.com/com-lihaoyi/mill/releases/download/0.9.8/0.9.8 > /usr/local/bin/mill && chmod +x /usr/local/bin/mill"`

liliwenaaa, 2021-07-16:
> It turns out that official link of mill is unavailable.
> You may try the following command:
> `sudo sh -c "curl -L https://github.com/com-lihaoyi/mill/releases/download/0.9.8/0.9.8 > /usr/local/bin/mill && chmod +x /usr/local/bin/mill"`

This is useful

closed 2021-07-19 by liliwenaaa