OpenXiangShan/XiangShan #883: make emu失败

QianpengLi577, 2021-07-15:
运行代码 `make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_THREADS=2 -j10`

出现如下错误

```
Done elaborating.
1 targets failed
XiangShan.test.runMain subprocess failed
java.io.IOException: com.sun.jna.LastErrorException: [104] 连接被对方重设
	at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.doRead(UnixDomainSocket.java:151)
	at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.read(UnixDomainSocket.java:122)
	at mill.main.client.ProxyStreamPumper.run(ProxyStreamPumper.java:23)
	at java.base/java.lang.Thread.run(Thread.java:829)
Caused by: com.sun.jna.LastErrorException: [104] 连接被对方重设
	at org.scalasbt.ipcsocket.UnixDomainSocketLibrary.read(Native Method)
	at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.doRead(UnixDomainSocket.java:149)
	... 3 more
make: *** [Makefile:82：build/SimTop.v] 错误 1
```

Makefile:78～94为

```
$(SIM_TOP_V): $(SCALA_FILE) $(TEST_FILE)
        mkdir -p $(@D)
        @echo "\n[mill] Generating Verilog files..." > $(TIMELOG)
        @date -R | tee -a $(TIMELOG)
        $(TIME_CMD) mill XiangShan.test.runMain $(SIMTOP) -td $(@D)      \
                --config $(CONFIG) --full-stacktrace --output-file $(@F)     \
                --infer-rw --repl-seq-mem -c:$(SIMTOP):-o:$(@D)/$(@F).conf   \
                --num-cores $(NUM_CORES) $(SIM_ARGS)
        $(MEM_GEN) $(@D)/$(@F).conf --output_file $(@D)/$(@F).sram.v
        @git log -n 1 >> .__head__
        @git diff >> .__diff__
        @sed -i 's/^/\/\// ' .__head__
        @sed -i 's/^/\/\//' .__diff__
        @cat .__head__ .__diff__ $@ $(@D)/$(@F).sram.v > .__out__
        @mv .__out__ $@
        @rm .__head__ .__diff__
        sed -i -e 's/$$fatal/xs_assert(`__LINE__)/g' $(SIM_TOP_V)
```

第82行为：        

`$(TIME_CMD) mill XiangShan.test.runMain $(SIMTOP) -td $(@D)      \`

java -version
```
openjdk version "11.0.11" 2021-04-20
OpenJDK Runtime Environment (build 11.0.11+9-Ubuntu-0ubuntu2.20.04)
OpenJDK 64-Bit Server VM (build 11.0.11+9-Ubuntu-0ubuntu2.20.04, mixed mode, sharing)
```

我百度了一下，或许是因为我的java版本不匹配？请问这个错误该怎么解决呢？

poemonsense, 2021-07-16:
这个好像是 mill 的问题，有的时候会出现跑完之后断开的情况。可以看一下 build 目录，应该已经生成出来了

QianpengLi577, 2021-07-16:
好嘞，不过现在build目录下只有这两个文件

> lock-emu                

> time.log

我再运行几次试试看~

sequencer, 2021-07-17:
Use `mill -I` to avoid this issue.

sunnybird, 2021-07-17:
modify the Makefile
diff --git a/Makefile b/Makefile
index e2b3253d..f5633038 100644
--- a/Makefile
+++ b/Makefile
@@ -79,7 +79,7 @@ $(SIM_TOP_V): $(SCALA_FILE) $(TEST_FILE)
        mkdir -p $(@D)
        @echo "\n[mill] Generating Verilog files..." > $(TIMELOG)
        @date -R | tee -a $(TIMELOG)
-       $(TIME_CMD) mill XiangShan.test.runMain $(SIMTOP) -td $(@D)      \
+       $(TIME_CMD) mill -i XiangShan.test.runMain $(SIMTOP) -td $(@D)      \
                --config $(CONFIG) --full-stacktrace --output-file $(@F)     \
                --infer-rw --repl-seq-mem -c:$(SIMTOP):-o:$(@D)/$(@F).conf   \
                --num-cores $(NUM_CORES) $(SIM_ARGS)

QianpengLi577, 2021-07-20:
> 这个好像是 mill 的问题，有的时候会出现跑完之后断开的情况。可以看一下 build 目录，应该已经生成出来了

我又试了几次，报错缺失一些头文件
```
/home/liqp/src/XS/src/test/csrc/common/device.cpp:17:10: fatal error: SDL2/SDL.h: 没有那个文件或目录
   17 | #include <SDL2/SDL.h>
      |          ^~~~~~~~~~~~
compilation terminated.
/home/liqp/src/XS/src/test/csrc/common/vga.cpp:18:10: fatal error: SDL2/SDL.h: 没有那个文件或目录
   18 | #include <SDL2/SDL.h>
      |          ^~~~~~~~~~~~
compilation terminated.
In file included from /home/liqp/src/XS/src/test/csrc/common/compress.cpp:16:
/home/liqp/src/XS/src/test/csrc/common/compress.h:21:10: fatal error: zlib.h: 没有那个文件或目录
   21 | #include <zlib.h>
      |          ^~~~~~~~
compilation terminated.
make[2]: *** [VSimTop.mk:84：compress.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[2]: *** [VSimTop.mk:86：device.o] 错误 1
make[2]: *** [VSimTop.mk:98：vga.o] 错误 1
/home/liqp/src/XS/src/test/csrc/common/keyboard.cpp:18:10: fatal error: SDL2/SDL.h: 没有那个文件或目录
   18 | #include <SDL2/SDL.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [VSimTop.mk:90：keyboard.o] 错误 1
In file included from /home/liqp/src/XS/src/test/csrc/common/ram.cpp:20:
/home/liqp/src/XS/src/test/csrc/common/compress.h:21:10: fatal error: zlib.h: 没有那个文件或目录
   21 | #include <zlib.h>
      |          ^~~~~~~~
compilation terminated.
make[2]: *** [VSimTop.mk:92：ram.o] 错误 1
/home/liqp/src/XS/src/test/csrc/common/sdcard.cpp: In function ‘void sd_read(uint32_t*)’:
/home/liqp/src/XS/src/test/csrc/common/sdcard.cpp:30:8: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
   30 |   fread(data, 4, 1, fp);
      |   ~~~~~^~~~~~~~~~~~~~~~
/home/liqp/src/XS/src/test/csrc/common/uart.cpp:32:12: warning: ‘int uart_dequeue()’ defined but not used [-Wunused-function]
   32 | static int uart_dequeue(void) {
      |            ^~~~~~~~~~~~
make[2]: 离开目录“/home/liqp/src/XS/build/emu-compile”
make[1]: *** [verilator.mk:118：build_emu_local] 错误 2
make[1]: 离开目录“/home/liqp/src/XS”
make: *** [verilator.mk:125：build/emu] 错误 2
```