OpenXiangShan/XiangShan #886: %Error: build/SimTop.v:2487721:3: Cannot find file containing module: 'array_18_ext'

liliwenaaa, 2021-07-17:
ubuntu 16.04
make emu
Sat, 17 Jul 2021 11:07:13 +0800
time -a -o ./build/time.log verilator --cc --exe --top-module SimTop +define+VERILATOR=1 +define+PRINTF_COND=1 +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_DELAY=0 -Wno-STMTDLY -Wno-WIDTH -I/opt/home/llw/tmp/xiangshan/XiangShan/build --x-assign unique -O3 -CFLAGS "-std=c++11 -static -Wall -I/opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/verilator -I/opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common -I/opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/difftest -DVERILATOR -Wno-maybe-uninitialized -DNUM_CORES=1" -LDFLAGS "-lpthread -lSDL2 -ldl -lz" --assert --stats-vars --output-split 30000 --output-split-cfuncs 30000 \
        -o /opt/home/llw/tmp/xiangshan/XiangShan/build/emu -Mdir build/emu-compile build/SimTop.v ./src/test/vsrc/common/difftest.v ./src/test/vsrc/common/ref.v ./src/test/vsrc/common/assert.v ./src/test/vsrc/common/ram.v /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/verilator/emu.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/verilator/main.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/verilator/snapshot.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/ram.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/uart.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/keyboard.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/vga.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/compress.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/device.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/flash.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/sdcard.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/common.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/common/axi4.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/difftest/interface.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/difftest/difftest.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/difftest/goldenmem.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/difftest/ref.cpp /opt/home/llw/tmp/xiangshan/XiangShan/src/test/csrc/difftest/nemuproxy.cpp
%Error: build/SimTop.v:2487721:3: Cannot find file containing module: 'array_18_ext'
2487721 |   array_18_ext array_18_ext (
        |   ^~~~~~~~~~~~
        ... Looked in:
             /opt/home/llw/tmp/xiangshan/XiangShan/build/array_18_ext
             /opt/home/llw/tmp/xiangshan/XiangShan/build/array_18_ext.v
             /opt/home/llw/tmp/xiangshan/XiangShan/build/array_18_ext.sv
             array_18_ext
             array_18_ext.v
             array_18_ext.sv
             build/emu-compile/array_18_ext
             build/emu-compile/array_18_ext.v
             build/emu-compile/array_18_ext.sv
%Error: build/SimTop.v:2487665:3: Cannot find file containing module: 'array_17_ext'
2487665 |   array_17_ext array_17_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487167:3: Cannot find file containing module: 'array_8_ext'
2487167 |   array_8_ext array_8_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2487008:3: Cannot find file containing module: 'array_7_ext'
2487008 |   array_7_ext array_7_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2486903:3: Cannot find file containing module: 'array_6_ext'
2486903 |   array_6_ext array_6_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2486798:3: Cannot find file containing module: 'array_5_ext'
2486798 |   array_5_ext array_5_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2486693:3: Cannot find file containing module: 'array_4_ext'
2486693 |   array_4_ext array_4_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2486588:3: Cannot find file containing module: 'array_3_ext'
2486588 |   array_3_ext array_3_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2487627:3: Cannot find file containing module: 'array_16_ext'
2487627 |   array_16_ext array_16_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487282:3: Cannot find file containing module: 'array_10_ext'
2487282 |   array_10_ext array_10_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487248:3: Cannot find file containing module: 'array_9_ext'
2487248 |   array_9_ext array_9_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2486483:3: Cannot find file containing module: 'array_2_ext'
2486483 |   array_2_ext array_2_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2486399:3: Cannot find file containing module: 'array_1_ext'
2486399 |   array_1_ext array_1_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2486337:3: Cannot find file containing module: 'array_0_ext'
2486337 |   array_0_ext array_0_ext (
        |   ^~~~~~~~~~~
%Error: build/SimTop.v:2486164:3: Cannot find file containing module: 'array_ext'
2486164 |   array_ext array_ext (
        |   ^~~~~~~~~
%Error: build/SimTop.v:2487834:3: Cannot find file containing module: 'array_21_ext'
2487834 |   array_21_ext array_21_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487798:3: Cannot find file containing module: 'array_20_ext'
2487798 |   array_20_ext array_20_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487762:3: Cannot find file containing module: 'array_19_ext'
2487762 |   array_19_ext array_19_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487554:3: Cannot find file containing module: 'array_15_ext'
2487554 |   array_15_ext array_15_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487461:3: Cannot find file containing module: 'array_14_ext'
2487461 |   array_14_ext array_14_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487396:3: Cannot find file containing module: 'array_13_ext'
2487396 |   array_13_ext array_13_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487340:3: Cannot find file containing module: 'array_12_ext'
2487340 |   array_12_ext array_12_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487311:3: Cannot find file containing module: 'array_11_ext'
2487311 |   array_11_ext array_11_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487961:3: Cannot find file containing module: 'array_24_ext'
2487961 |   array_24_ext array_24_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487923:3: Cannot find file containing module: 'array_23_ext'
2487923 |   array_23_ext array_23_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2487867:3: Cannot find file containing module: 'array_22_ext'
2487867 |   array_22_ext array_22_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2488055:3: Cannot find file containing module: 'array_26_ext'
2488055 |   array_26_ext array_26_ext (
        |   ^~~~~~~~~~~~
%Error: build/SimTop.v:2488017:3: Cannot find file containing module: 'array_25_ext'
2488017 |   array_25_ext array_25_ext (
        |   ^~~~~~~~~~~~
%Error: Exiting due to 28 error(s)
verilator.mk:104: recipe for target 'build/emu-compile/VSimTop.mk' failed
make: *** [build/emu-compile/VSimTop.mk] Error 1

poemonsense, 2021-07-17:
For SRAMs, currently Chisel generates SRAM via verilog blackboxes. A python script is used to generate these SRAM modules with names starting with `array_*`. This step is done after the mill/scala process. We will remove the script after Chisel updates the transforms to include `vlsi_mem_gen` (issue #858). 

`Cannot find file containing module` happens when `make` stops before the script runs due to some errors, in which case the SRAM modules are not generated.

To solve this, please try to manually run the following command, which is also in the `Makfile`
```
./scripts/vlsi_mem_gen build/SimTop.v.conf --output_file build/SimTop.v.sram.v
cat build/SimTop.v build/SimTop.v.sram.v > .__out__
mv .__out__ build/SimTop.v
```

If the above does not work, please try
```
rm -rf build
# make again here
```

liliwenaaa, 2021-07-17:
> For SRAMs, currently Chisel generates SRAM via verilog blackboxes. A python script is used to generate these SRAM modules with names starting with `array_*`. This step is done after the mill/scala [...]
get error
./scripts/vlsi_mem_gen build/SimTop.v.conf --output_file build/SimTop.v.sram.v
  File "./scripts/vlsi_mem_gen", line 28
    self.port_spec.append(f'{io_type} {width_str}{name}')
                                                       ^
SyntaxError: invalid syntax

sequencer, 2021-07-17:
Google plz: fstring

closed 2021-07-19 by liliwenaaa

liliwenaaa, 2021-07-19:
Problem in my python  version，pyhton3.6 replace python3.5