OpenXiangShan/XiangShan #891: make emu 失败

bianliangpai, 2021-07-19:
环境为 ubuntu 18.04，按照 readme.md 中，先 git clone，再 make init，再 make verilog，反复尝试多次均会报错。
没有出现 java exception，看起来与现有的其他 issue 均不是同一个问题。

输出如下：
`mkdir -p build
mill XiangShan.runMain top.TopMain -td build                      \
	--config DefaultConfig --full-stacktrace --output-file XSTop.v     \
	--disable-all --remove-assert --infer-rw                     \
	--repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf  \
	--num-cores 1
[191/191] XiangShan.runMain 
Elaborating design...
FPGASoC cores: 1 banks: 4 block size: 64 bus size: 256
INT Regfile: 14R8W
FP  Regfile: 14R8W
RS 0: type AluExeUnit, size 64, enq 4, deq 4, numSrc 2, fast 8, wakeup 8
RS 1: type MulDivExeUnit, size 32, enq 4, deq 2, numSrc 2, fast 6, wakeup 8
RS 2: type JmpExeUnit, size 16, enq 1, deq 1, numSrc 2, fast 6, wakeup 8
RS 3: type FmacExeUnit, size 64, enq 4, deq 4, numSrc 3, fast 4, wakeup 8
RS 4: type FmiscExeUnit, size 32, enq 4, deq 2, numSrc 2, fast 4, wakeup 8
RS 5: type LoadExu, size 16, enq 1, deq 1, numSrc 1, fast 8, wakeup 8
RS 6: type LoadExu, size 16, enq 1, deq 1, numSrc 1, fast 8, wakeup 8
RS 7: type StoreExu, size 16, enq 1, deq 1, numSrc 2, fast 6, wakeup 16
RS 8: type StoreExu, size 16, enq 1, deq 1, numSrc 2, fast 6, wakeup 16
Scheduler: 
  number of issue ports: 17
  number of replay ports: 4
  number of std ports: 2
  number of outside fast wakeup ports: 2
[WARN] Signal |DISPLAY_LOG_ENABLE| has multiple sinks
[WARN] Signal |logTimestamp| has multiple sinks
ImmUnion max len: 20
LoadQueue: size:64
StoreQueue: size:48
Regfile: size:160 read: 14write: 8
Regfile: size:160 read: 14write: 8
Fast wakeup: RS 0 -> List(0, 1, 2, 5, 6, 7, 8), source: [0,4)
Fast wakeup: RS 1 -> List(0, 1, 2, 5, 6, 7, 8), source: [6,8)
Fast wakeup: RS 3 -> List(3, 4), source [8, 12)
Fast wakeup: RS 5 -> List(0, 5, 6), source: [4,5)
Fast wakeup: RS 6 -> List(0, 5, 6), source: [5,6)
FPGAPlatform:true EnableDebug:true

Memory map:
[h00_0000_0000 -> h00_0FFF_FFFF] Width:unlimited Description:Reserved [RW]
[h00_1000_0000 -> h00_1FFF_FFFF] Width:unlimited Description:QSPI_Flash [RWX]
[h00_2000_0000 -> h00_2FFF_FFFF] Width:unlimited Description:Reserved [RW]
[h00_3000_0000 -> h00_3000_FFFF] Width:unlimited Description:DMA [RW]
[h00_3001_0000 -> h00_3004_FFFF] Width:unlimited Description:GPU [RWC]
[h00_3005_0000 -> h00_3006_FFFF] Width:unlimited Description:USB/SDMMC [RW]
[h00_3007_0000 -> h00_30FF_FFFF] Width:unlimited Description:Reserved [RW]
[h00_3100_0000 -> h00_3111_FFFF] Width:unlimited Description:MMIO [RW]
[h00_3112_0000 -> h00_37FF_FFFF] Width:unlimited Description:Reserved [RW]
[h00_3800_0000 -> h00_3800_FFFF] Width:unlimited Description:CLINT [RW]
[h00_3801_0000 -> h00_3801_FFFF] Width:unlimited Description:BEU [RW]
[h00_3802_0000 -> h00_3BFF_FFFF] Width:unlimited Description:Reserved []
[h00_3C00_0000 -> h00_3FFF_FFFF] Width:unlimited Description:PLIC [RW]
[h00_4000_0000 -> h00_7FFF_FFFF] Width:unlimited Description:PCIe [RW]
[h00_8000_0000 -> h1F_FFFF_FFFF] Width:unlimited Description:DDR [RWXIDSA]

Ftq: size:48
Roq: size:192 wbports:18 commitwidth:6
CSR: hasEmuPerfCnt:false

int wb arbiter:
[ AluExeUnit ] -> out #0
[ AluExeUnit ] -> out #1
[ AluExeUnit ] -> out #2
[ AluExeUnit ] -> out #3
[ LoadExu ] -> out #4
[ LoadExu ] -> out #5
[ MulDivExeUnit JmpExeUnit ] -> arb -> out #6
[ MulDivExeUnit FmiscExeUnit FmiscExeUnit ] -> arb -> out #7

1 targets failed
XiangShan.runMain subprocess failed
Makefile:49: recipe for target 'build/XSTop.v' failed
make: *** [build/XSTop.v] Error 1`

poemonsense, 2021-07-19:
机器内存有多大？

bianliangpai, 2021-07-19:
> 机器内存有多大？

可用内存 15G 左右，以下是 top 的输出
KiB Mem : 16369580 total, 15420432 free,   733028 used,   216120 buff/cache

poemonsense, 2021-07-20:
试试把build.sc里面的两个64G都换成15G，看看结果会怎么样？

bianliangpai, 2021-07-20:
我把 build.sc 里的两个 64G 改成了 15G\12G，但内存依然被耗光了。这次全程盯着 top 输出看，确实是在可用内存降到 100M 以下的时候退出的。这样就确定是内存不足导致的问题了。
我又把 build.sc 里的配置改成了 10G，实际占用大致为 15G 左右。由于实际占用存在上下波动，我的这次构建在内存占用向上波动时又失败了。我个人感觉 build.sc 里配置的大小并不是直接限制了占用的数值，不知道是不是一个已知现象？
最后，我把主机（虚拟机）的内存调到了24G，终于成功了。

closed 2021-07-21 by bianliangpai

MapleLeafFall, 2021-11-24:
感谢参考值……
我给ubuntu20.04分配30G内存，build.sc的参数改为10G也可以了 :)

liushi089039, 2023-06-07:
riscv64-linux-gnu-gcc：命令未找到

wwwuxy, 2024-01-15:
> riscv64-linux-gnu-gcc：命令未找到
metoo

chenguokai, 2024-01-15:
> riscv64-linux-gnu-gcc：命令未找到
> metoo

No risc-v cross toolchain should be required when `make verilog`. 
It seems to be irrelevant to this issue.  Please open another issue if you believe this is triggered in `make verilog` process.

yaoyuexiaogege, 2024-10-17:
mkdir -p build/rtl
time -avp -0 ./build/time.log mill -i xiangshan.runMain top.TopMain
seq-mem --repl-seq-mem-file=XSTop.sv.conf"
--target-dir build/rtl --config DefaultConfig --issue B--firtool-opt "--repl
--num-cores 1 --dump-fir --target systemverilog--split-verilog - firtool-opt
"-O=release - - disable-annotation-unknown - lowering-options=explicitBitcast,disallowLo
calVariables,disallowPortDeclSharing,locationInfoStyle=none" --fpga-platform --disable
-all --remove-assert --reset-gen --disable-always-basic-diff
/scripts/gen_sep_mem.sh "./scripts/vlsi_mem_gen" "buuild/rtl/XSTop.sv.conf" build/rtl
cat: build/rtl/XSTop.sv.conf: No such file or directory
rm: cannot remove 'build/rtl/XSTop.sv.conf.tmp': No such file or directory
make: *** [Makefile:155: build/rtl/XSTop.sv] Erгог 1
运行make verilog后报错，请问我这是什么原因？

WenZhengkai, 2024-10-20:
> mkdir -p build/rtl time -avp -0 ./build/time.log mill -i xiangshan.runMain top.TopMain seq-mem --repl-seq-mem-file=XSTop.sv.conf" --target-dir build/rtl --config DefaultConfig --issue B--firtool-opt [...]

我遇到过相同的问题，我的原因setup-tool.sh脚本运行过程中，由于网络波动，mill相关文件未安装成功。解决网络问题后，重新运行相关安装脚本，解决了该问题。

yaoyuexiaogege, 2024-10-23:
> 感谢参考值…… 我给ubuntu20.04分配30G内存，build.sc 的参数改为10G也可以了 :)

你好，想问下，如何修改build.sc，是改哪一行代码？感觉里面没有描述内存大小的呀