OpenXiangShan/XiangShan #894: ready-to-run/riscv64-nemu-interpreter-so cannot be loaded

bianliangpai, 2021-07-20:
环境 
ubuntu 18.04 

输出
$ ./build/emu -b 0 -e 0 -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so

Emu compiled at Jul 20 2021, 12:03:41
The image is ./ready-to-run/coremark-2-iteration.bin
Using simulated 8192MB RAM
[warning] sdcard img not found
Using ./ready-to-run/riscv64-nemu-interpreter-so for difftest
emu: /path/to/XiangShan/src/test/csrc/difftest/nemuproxy.cpp:50: NemuProxy::NemuProxy(int): Assertion `handle' failed.
[1]    122576 abort (core dumped)  ./build/emu -b 0 -e 0 -i ./ready-to-run/coremark-2-iteration.bin --diff

检查编译结果，该文件存在
ls -al /path/to/XiangShan/ready-to-run/riscv64-nemu-interpreter-so              
-rwxrwxr-x 1 {user} {user} 2076568 Jul 19 14:19 /path/to/XiangShan/ready-to-run/riscv64-nemu-interpreter-so

poemonsense, 2021-07-21:
可以试一下手动编译 nemu-so

在 NEMU 仓库中 `make ISA=riscv64 SHARE=1 -j8`，结果在 `build` 目录下面

liliwenaaa, 2021-07-21:
手动编译NEMU可行

poemonsense, 2021-07-21:
可能是因为库版本不一致，导致默认自带的那个so跑不起来

bianliangpai, 2021-07-21:
确实，加载手动编译的库就可以了。谢谢。

closed 2021-07-21 by bianliangpai