OpenXiangShan/XiangShan #897: env配置的仓库路径不对

zhuxuanlong, 2021-07-24:
rocket-chip下面的torture配置的env路径是：
[submodule "env"]
	path = env
	url = git://github.com/ucb-bar/riscv-test-env.git

而env实际上是在riscv仓库下面，导致git init下载失败。

poemonsense, 2021-07-24:
make init没有加recursive，不需要下载rocket-chip下面的其他submodule。

这个问题我们已经在 #870 中说过了

closed 2021-07-24 by poemonsense

zhuxuanlong, 2021-07-24:
那么就出现另外一个问题，我昨天新下载代码，make init，并没有提示需要更新什么了，但执行make emu报错：
root@zte-R5300-G4-riscv:/share/zhuxuanlong/xiangshan_master/XiangShan# make init
git submodule update --init
root@zte-R5300-G4-riscv:/share/zhuxuanlong/xiangshan_master/XiangShan#

root@zte-R5300-G4-riscv:/share/zhuxuanlong/xiangshan_master/XiangShan# make emu SIM_ARGS=--disable-log EMU_THREADS=16
mkdir -p build
Sat, 24 Jul 2021 02:26:13 +0800
time -a -o ./build/time.log mill XiangShan.test.runMain top.SimTop -td build      \
        --config DefaultConfig --full-stacktrace --output-file SimTop.v     \
        --infer-rw --repl-seq-mem -c:top.SimTop:-o:build/SimTop.v.conf   \
        --num-cores 1 --disable-log

Compiling /share/zhuxuanlong/xiangshan_master/XiangShan/build.sc
[27/213] api-config-chipsalliance.compile
Compiling compiler interface...
[info] compiling 1 Scala source to /share/zhuxuanlong/xiangshan_master/XiangShan/out/api-config-chipsalliance/compile/dest/classes ...
[info] done compiling
[51/213] rocket-chip.macros.compile
[info] compiling 1 Scala source to /share/zhuxuanlong/xiangshan_master/XiangShan/out/rocket-chip/macros/compile/dest/classes ...
[info] done compiling
[104/213] rocket-chip.compile
[info] compiling 376 Scala sources to /share/zhuxuanlong/xiangshan_master/XiangShan/out/rocket-chip/compile/dest/classes ...
[error] /share/zhuxuanlong/xiangshan_master/XiangShan/rocket-chip/src/main/scala/tile/FPU.scala:293:25: not found: value hardfloat
[error]   def recode(x: UInt) = hardfloat.recFNFromFN(exp, sig, x)
[error]                         ^

我看代码，确实伯克利硬浮点没有下全。

也就是碰到这种问题，如何进行？