OpenXiangShan/XiangShan #906: OpenJDK 64-Bit Server问题

dreamflyings, 2021-07-28:
make verilog/make emu出现如下问题：
Connection:[DISPLAY_LOG_ENABLE] type:[func] source location:[SimTop] sink location:[AXI4RAM_1]
Connection:[logTimestamp] type:[func] source location:[SimTop] sink location:[AXI4RAM_1]
Connection:[XSPERF_CLEAN] type:[func] source location:[SimTop] sink location:[L2Prefetcher]
Connection:[XSPERF_DUMP] type:[func] source location:[SimTop] sink location:[L2Prefetcher]
[deprecated] ListBuffer.scala:153 (5 calls): do_toBools is deprecated: "Use asBools instead"
[deprecated] ListBuffer.scala:164 (5 calls): do_toBools is deprecated: "Use asBools instead"
[warn] There were 2 deprecated function(s) used. These may stop compiling in a future release - you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[warn]   In the sbt interactive console, enter:
[warn]     set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[warn]   or, in your build.sbt, add the line:
[warn]     scalacOptions := Seq("-unchecked", "-deprecation")
Done elaborating.
#
# There is insufficient memory for the Java Runtime Environment to continue.
# Native memory allocation (mmap) failed to ma 6962544640 bytes for committing reserved memory.
# An error report file with more information is saved as:
# /home/usrname/Projec!ts/XiangShan/hs_err_pid26575.log
OpenJDK 64-Bit Server VM warning: INFO: os::commit_mmory(0x00007f4e9d000000, 6962544640, 0) failed; error='Not enough space' (�errno=12)
�1 targets Makefile:79: recipe for target 'build/SimTop.v' failed
make: *** [build/SimTop.v] Error 1
_______________________________________________________________________________________
如果将
./build.sc:84:  override def forkArgs = Seq("-Xmx64G")
./build.sc:95:    override def forkArgs = Seq("-Xmx64G")
./out/XiangShan/test/forkArgs/meta.json:3:        "-Xmx64G"
./out/XiangShan/forkArgs/meta.json:3:        "-Xmx64G"
这些文件中的jvm_args: -Xmx64G 改小点则可以通过，希望这个参数用户可配
正确的log文件如下：
[warn] There were 2 deprecated function(s) used. These may stop compiling in a future release - you are encouraged to fix these issues.
[warn] Line numbers for deprecations reported by Chisel may be inaccurate; enable scalac compiler deprecation warnings via either of the following methods:
[warn]   In the sbt interactive console, enter:
[warn]     set scalacOptions in ThisBuild ++= Seq("-unchecked", "-deprecation")
[warn]   or, in your build.sbt, add the line:
[warn]     scalacOptions := Seq("-unchecked", "-deprecation")
Done elaborating.

（＊好久＊）

xiangshan.backend.MemBlock@102db955
java.io.IOException: com.sun.jna.LastErrorException: [104] 连接被对方重设
./scripts/vlsi_mem_gen build/XSTop.v.conf --tsmc28 --output_file build/tsmc28_sram.v > build/tsmc28_sram.v.conf
./scripts/vlsi_mem_gen build/XSTop.v.conf --output_file build/sim_sram.v
# sed -i -e 's/_\(aw\|ar\|w\|r\|b\)_\(\|bits_\)/_\1/g' build/XSTop.v

poemonsense, 2021-07-28:
可以手动改一下 `build.sc`，每个机器的环境不同，我们不好给一个最小值

fatalfeel, 2021-08-19:
gedit /root/riscv_cpu/XiangShan/build.sc
#find all
-Xmx64G
#change to
-Xmx16G

#generate .v
make verilog -j8

closed 2023-02-15 by AugustusWillisWang