OpenXiangShan/XiangShan #915: master 分支在加入difftest子文件夹代码后，build.sbt文件希望添加对应的依赖，这样sbt构建环境简单修改Makefile就可支持了，毕竟有些需要经过代理的网络mill工具配置还是比较麻烦的，谢谢！

dongdeji, 2021-08-16:
下面是修改后可以支持sbt构建的，可以参考，谢谢！
xiangshan_code_root_folder# git diff build.sbt
diff --git a/build.sbt b/build.sbt
index f6a092c..1421e3c
--- a/build.sbt
+++ b/build.sbt
@@ -45,9 +45,12 @@ lazy val `block-inclusive-cache` = (project in file("block-inclusivecache-sifive
   )
   .dependsOn(`rocket-chip`)

+lazy val difftest = (project in file("difftest"))
+  .settings(commonSettings, chiselSettings)
+
 lazy val chiseltest = (project in file("chiseltest"))
   .settings(commonSettings, chiselSettings)

 lazy val xiangshan = (Project("XiangShan", base = file(".")))
   .settings(commonSettings, chiselSettings)
-  .dependsOn(`rocket-chip`, `block-inclusive-cache`, chiseltest)
\ No newline at end of file
+  .dependsOn(`rocket-chip`, `block-inclusive-cache`, chiseltest, difftest)
\ No newline at end of file

dongdeji, 2021-08-16:
另外最好将的xiangshan_code_root_folder/project/build.properties中的sbt.version改成1.5.4，即sbt.version=1.5.4

poemonsense, 2021-08-17:
欢迎提pull request

dongdeji, 2021-08-25:
OK！

closed 2021-08-25 by dongdeji