OpenXiangShan/XiangShan #931: full install step share

fatalfeel, 2021-08-19:
#risc-v cpu full project, I use root login ubuntu 16.04
apt install verilog
sh -c "curl -L https://github.com/com-lihaoyi/mill/releases/download/0.9.9/0.9.9 > /usr/local/bin/mill && chmod +x /usr/local/bin/mill"
mill version

mkdir -p /root/riscv_cpu
cd /root/riscv_cpu

git clone --recursive https://github.com/OpenXiangShan/NEMU.git
git clone --recursive https://github.com/OpenXiangShan/nexus-am.git
git clone --recursive https://github.com/OpenXiangShan/XiangShan.git
git clone --recursive https://github.com/verilator/verilator

#absolute path
export NEMU_HOME="/root/riscv_cpu/NEMU"
export AM_HOME="/root/riscv_cpu/nexus-am"
export NOOP_HOME="/root/riscv_cpu/XiangShan"

cd verilator
autoconf
./configure
make -j8 && make install

cd /root/riscv_cpu/XiangShan
make init

gedit /root/riscv_cpu/XiangShan/build.sc
#find all
-Xmx64G
#change to
-Xmx16G

#generate .v
make verilog -j8

#simulation
make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_THREADS=2 -j8
./build/emu -b 0 -e 0 -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so

poemonsense, 2021-08-20:
Thanks.

We recommend using stable verison verilator, like v4.204. Verilator v4.210 has performance issues on XiangShan.

fatalfeel, 2021-08-20:
the -Xmx64G for the most people, will build failed. better using -Xmx16G

poemonsense, 2021-08-20:
We need a larger stack for the normal XiangShan, which 16G tends to be not enough. So we prefer to keep it large and sorry for the inconvenience.

fatalfeel, 2021-08-20:
not enough for compile or simulation?
i use 16G compile and simulation successful
root@linuxpc:~/riscv_cpu/XiangShan# ./build/emu -b 0 -e 0 -i ./ready-to-run/coremark-2-iteration.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so
Emu compiled at Aug 19 2021, 23:21:19
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 23334
The image is ./ready-to-run/coremark-2-iteration.bin
Using simulated 8192MB RAM
[warning] sdcard img not found
Using ./ready-to-run/riscv64-nemu-interpreter-so for difftest
The first instruction of core 0 has commited. Difftest enabled. 
Running CoreMark for 2 iterations
^CCore 0: SOME SIGNAL STOPS THE PROGRAM at pc = 0x80000ff8
total guest instructions = 29,2743
instrCnt = 29,2743, cycleCnt = 26,1398, IPC = 1.119913
Seed=0 Guest cycle spent: 26,1400 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 15,0658ms

poemonsense, 2021-08-20:
`make emu CONFIG=MinimalConfig` produces minimal XS

fatalfeel, 2021-08-20:
#for IntelliJ IDE
cd /root/riscv_cpu/XiangShan
mill mill.scalalib.GenIdea/idea
run /opt/idea-IC-193.7288.26/bin/idea.sh
open directory /root/riscv_cpu/XiangShan
File -> Project Structure -> Project Settings -> Project
Project SDK: 1.8 (your java version)
Build -> Build Project

right click /root/riscv_cpu/XiangShan/src/test/scala/top/SimTop.scala
[Edit] -> Configuration -> Environment variables: _JAVA_OPTIONS=-Xmx32G

right click /root/riscv_cpu/XiangShan/src/test/scala/top/SimTop.scala
[Debug]

fatalfeel, 2021-08-24:
i will keep this open for first building user, if sir Poemonsense need to close u can do it by urself

LIKEJMJJ, 2021-09-06:
> #risc-v cpu全项目，我用root登录ubuntu 16.04
> apt install verilog
> sh -c "curl -L https://github.com/com-lihaoyi/mill/releases/download/0.9.9/0.9.9 > /usr/local/bin/mill && chmod +x /usr/local/bin/mill"
> 磨机版本
> 
> mkdir -p /root/riscv_cpu
> cd /root/riscv_cpu
> 
> git clone --recursive https://github.com/OpenXiangShan/NEMU.git
> git clone --recursive https://github.com/OpenXiangShan/nexus-am.git
> git clone --recursive [https://github.com /OpenXiangShan/XiangShan.git](https://github.com/OpenXiangShan/XiangShan.git)
> git clone --recursive https://github.com/verilator/verilator
> 
> #绝对路径
> export NEMU_HOME="/root/riscv_cpu/NEMU"
> export AM_HOME="/root/riscv_cpu/nexus-am"
> export NOOP_HOME="/root/riscv_cpu/XiangShan"
> 
> cd verilator
> autoconf
> ./configure
> make -j8 && make install
> 
> cd /root/riscv_cpu/XiangShan
> make init
> 
> gedit /root/riscv_cpu/XiangShan/build.sc
> #find all
> -Xmx64G
> #change to
> -Xmx16G
> (我的电脑是16g ram 16g swap ubuntu 16.04.7)
> 
> #generate .v
> make verilog -j8
> 
> #simulation
> make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_THREADS=2 -j8
> ./build/emu -b 0 -e 0 -i ./ready-to-run/coremark-2-iteration.bin --diff ./准备运行/riscv64-nemu-interpreter-so
> 
> 参考：[https](https://fatalfeel.blogspot.com/2013/12/chisel-design-ic-for-risc-v.html) : [//fatalfeel.blogspot.com/2013/12/chisel-design-ic-for-risc-v.html](https://fatalfeel.blogspot.com/2013/12/chisel-design-ic-for-risc-v.html)

请问一下你是怎么在Ubuntu装上g++-riscv64-linux-gnu的？

fatalfeel, 2021-09-06:
you do need build gcc in OpenXiangShan project, but if you need try this

1.
#preinstall
apt install autoconf automake autotools-dev curl libmpc-dev libmpfr-dev libgmp-dev libusb-1.0-0-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev device-tree-compiler pkg-config libexpat-dev libfl-dev

2.
#export must add to last line of ~/.bashrc
export RISCV="/opt/riscv"
export ROCKETCHIP="/root/riscv_cpu/rocket-chip"
export MAKEFLAGS="$MAKEFLAGS -j8"

3.
git clone --recursive https://github.com/freechipsproject/rocket-tools

4.
cd ~/riscv_cpu/rocket-tools
git submodule update --init --recursive
./build.sh

refer to: https://fatalfeel.blogspot.com/2013/12/rocket-chip-risc-v-cpu-core-debugging.html

poemonsense, 2021-09-06:
`sudo apt install g++-riscv64-linux-gnu`

xaoyaolee, 2022-06-03:
[error] /home/xaoyao/riscv-cpu/XiangShan/fudian/src/main/scala/fudian/Generator.scala:13:29: value getPackageName is not a member of Class[T]
[error]     val pkg = this.getClass.getPackageName
[error]                             ^
[error] one error found

closed 2023-10-21 by cebarobot