OpenXiangShan/XiangShan #932: can you make a version for IntelliJ

fatalfeel, 2021-08-20:
can you make a version for IntelliJ

poemonsense, 2021-08-20:
[mill supports IDEA](https://github.com/com-lihaoyi/mill#intellij-setup): run `mill mill.scalalib.GenIdea/idea` to generate IDEA project files

fatalfeel, 2021-08-20:
thanks man

poemonsense, 2021-08-20:
Thanks for the idea. Feel free to add it to readme and create a PR. I will merge it.

fatalfeel, 2021-08-21:
#for IntelliJ IDE
cd /root/riscv_cpu/XiangShan
mill mill.scalalib.GenIdea/idea
run /opt/idea-IC-193.7288.26/bin/idea.sh
open directory /root/riscv_cpu/XiangShan
File -> Project Structure -> Project Settings -> Project
Project SDK: 1.8 (your java version)
Build -> Build Project

right click /root/riscv_cpu/XiangShan/src/test/scala/top/SimTop.scala
[Edit] -> Configuration -> Environment variables: _JAVA_OPTIONS=-Xmx32G

right click /root/riscv_cpu/XiangShan/src/test/scala/top/SimTop.scala
[Debug]

closed 2021-08-23 by fatalfeel