OpenXiangShan/XiangShan #933: how to apply SimTop.v in Vivado

fatalfeel, 2021-08-20:
load source file /root/riscv_cpu/XiangShan/build/SimTop.v 
/root/riscv_cpu/XiangShan/build/plusarg_reader.v
or 
/root/riscv_cpu/XiangShan/build/XSTop.v
/root/riscv_cpu/XiangShan/build/plusarg_reader.v

make Schematic failed
[Synth 8-439] module 'plusarg_reader' not found ["/root/riscv_cpu/XiangShan/build/SimTop.v":622]
[Synth 8-6156] failed synthesizing module 'TLMonitor' ["/root/riscv_cpu/XiangShan/build/SimTop.v":35]
[Vivado_Tcl 4-5] Elaboration failed - please see the console for details

how do to use it correctly

poemonsense, 2021-08-20:
SimTop is for behavior simulation. If you need synthesizable verilog, please use `make verilog` to generate top module.

As for `module plusarg_reader not found`, please have a look at `build`, which should contain plusarg_reader.v and other verilog files.

fatalfeel, 2021-08-20:
thanks and is top module -> XSTop.v?

poemonsense, 2021-08-20:
Yes. It's XSTop, with 3 AXI channels

fatalfeel, 2021-08-20:
thanks a lot man

fatalfeel, 2021-08-21:
Done for this
#XiangShan for "vivado hl system edition" 2019.2.1
File -> Project -> New project -> RTL project
Project name: XSTop
Project location: ~/riscv_cpu/XiangShan/build
Create project subdirectory - uncheck
Add Sources:
XSTop.v
plusarg_reader.v
tsmc28_sram.v
Add Constraints: None
Select last part: xqzu29dr-ffrf1760-1M-m
#or
Select last board: Zynq UltraScale+ RFSoC ZCU1285 Characterization Kit
open view [Flow Navigator]
RTL ANALYSIS: Open Elaborated Design -> Schematic (see the circuit)

qinyunchuan, 2021-08-22:
@fatalfeel What's the synthesized frequency of  XSTop on your FPGA ?

poemonsense, 2021-08-22:
> Select last board: Zynq UltraScale+ RFSoC ZCU1285 Characterization Kit

Did you use MinimalConfig of XiangShan? We actually don't test enough on this minimal version. So there may be some functional or timing issues with this version.

fatalfeel, 2021-08-22:
I did not set synthesized frequency only draw a cpu circuit

closed 2021-08-23 by fatalfeel