OpenXiangShan/XiangShan #939: idea java.lang.StackOverflowError

fatalfeel, 2021-08-21:
When Idea in debug mode it will StackOverflowError
There are recursive happend, how do i modify those code

///
Exception in thread "main" java.lang.StackOverflowError
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:717)
	at firrtl.transforms.ConstantPropagation.$anonfun$constPropModule$14(ConstantPropagation.scala:714)
	at scala.collection.mutable.HashMap.getOrElseUpdate(HashMap.scala:86)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:714)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:717)
	at firrtl.transforms.ConstantPropagation.$anonfun$constPropModule$14(ConstantPropagation.scala:714)
	at scala.collection.mutable.HashMap.getOrElseUpdate(HashMap.scala:86)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:714)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:717)
	at firrtl.transforms.ConstantPropagation.$anonfun$constPropModule$14(ConstantPropagation.scala:714)
	at scala.collection.mutable.HashMap.getOrElseUpdate(HashMap.scala:86)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:714)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:717)
	at firrtl.transforms.ConstantPropagation.$anonfun$constPropModule$14(ConstantPropagation.scala:714)
	at scala.collection.mutable.HashMap.getOrElseUpdate(HashMap.scala:86)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:714)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:717)
	at firrtl.transforms.ConstantPropagation.$anonfun$constPropModule$14(ConstantPropagation.scala:714)
	at scala.collection.mutable.HashMap.getOrElseUpdate(HashMap.scala:86)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:714)
	at firrtl.transforms.ConstantPropagation.regConstant$1(ConstantPropagation.scala:717)
	at firrtl.transforms.ConstantPropagation.$anonfun$constPropModule$14(ConstantPropagation.scala:714)
	at scala.collection.mutable.HashMap.getOrElseUpdate(HashMap.scala:86)

poemonsense, 2021-08-22:
I think some mofications are needed in IDEA settings. However, we don't officially support using IDEA to run XiangShan (actully we internally only use IDEA for coding). Probably you need to search the web or ask IDEA guys on how to change the stack size.

fatalfeel, 2021-08-22:
Run will be pass and ok 
only in debug will be failed 
even add size in -Xss -Xmx still failed

If could  someday can implement in Idea