OpenXiangShan/XiangShan #940: xiangshan compile error

zhaoyukoon, 2021-08-22:
after `make init`, i try to run and get failure:
```
../mill/mill XiangShan.runMain top.TopMain -td build --config DefaultConfig --full-stacktrace --output-file XSTop.v   --disable-all --remove-assert --infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf --num-cores 4
```

i got following log:
```
Exception in thread "main" chisel3.internal.ChiselException: Connection between sink (Bool[1](IO io_hartIsInReset in TLDebugModule)) and source (Bool[4](IO in unelaborated DebugModule)) failed @: Sink and Source are different length Vecs.
        at chisel3.internal.throwException$.apply(Error.scala:155)
        at chisel3.Data.connect(Data.scala:416)
        at chisel3.Vec.$colon$eq(Aggregate.scala:237)
        at device.DebugModule$$anon$1.<init>(RocketDebugWrapper.scala:64)
        at device.DebugModule.module$lzycompute(RocketDebugWrapper.scala:55)
        at device.DebugModule.module(RocketDebugWrapper.scala:55)
        at device.DebugModule.module(RocketDebugWrapper.scala:41)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$2(LazyModule.scala:278)
        at chisel3.Module$.do_apply(Module.scala:54)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:278)
        at scala.collection.immutable.List.flatMap(List.scala:338)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:276)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:273)
        at freechips.rocketchip.diplomacy.LazyRawModuleImp.instantiate(LazyModule.scala:344)
        at freechips.rocketchip.diplomacy.LazyRawModuleImp.$anonfun$x$22$1(LazyModule.scala:357)
        at chisel3.withClockAndReset$.apply(MultiClock.scala:25)
        at freechips.rocketchip.diplomacy.LazyRawModuleImp.<init>(LazyModule.scala:357)
        at top.XSTopWithoutDMA$$anon$5.<init>(Top.scala:371)
        at top.XSTopWithoutDMA.module$lzycompute(Top.scala:371)
        at top.XSTopWithoutDMA.module(Top.scala:371)
        at top.TopMain$.$anonfun$main$1(Top.scala:454)
        at chisel3.Module$.do_apply(Module.scala:54)
        at chisel3.stage.ChiselGeneratorAnnotation.$anonfun$elaborate$1(ChiselAnnotations.scala:60)
        at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:656)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
        at chisel3.internal.Builder$.build(Builder.scala:653)
        at chisel3.internal.Builder$.build(Builder.scala:649)
        at chisel3.stage.ChiselGeneratorAnnotation.elaborate(ChiselAnnotations.scala:60)
        at chisel3.stage.phases.Elaborate.$anonfun$transform$1(Elaborate.scala:26)
        at scala.collection.TraversableLike.$anonfun$flatMap$1(TraversableLike.scala:245)
        at scala.collection.immutable.List.foreach(List.scala:392)
        at scala.collection.TraversableLike.flatMap(TraversableLike.scala:245)
        at scala.collection.TraversableLike.flatMap$(TraversableLike.scala:242)
        at scala.collection.immutable.List.flatMap(List.scala:355)
        at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:24)
        at chisel3.stage.phases.Elaborate.transform(Elaborate.scala:17)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
        at firrtl.options.Translator.transform(Phase.scala:248)
        at firrtl.options.Translator.transform$(Phase.scala:248)
        at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
        at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:278)
        at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
        at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
        at scala.collection.immutable.List.foldLeft(List.scala:89)
        at firrtl.options.DependencyManager.transform(DependencyManager.scala:269)
        at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
        at firrtl.options.PhaseManager.transform(DependencyManager.scala:436)
        at chisel3.stage.ChiselStage.run(ChiselStage.scala:45)
        at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
        at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
        at firrtl.options.Translator.transform(Phase.scala:248)
        at firrtl.options.Translator.transform$(Phase.scala:248)
        at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
        at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
        at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
        at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
        at scala.collection.immutable.List.foldLeft(List.scala:89)
        at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
        at logger.Logger$.$anonfun$makeScope$2(Logger.scala:166)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
        at logger.Logger$.makeScope(Logger.scala:164)
        at firrtl.options.Stage.transform(Stage.scala:47)
        at firrtl.options.Stage.execute(Stage.scala:58)
        at top.XiangShanStage$.execute(XiangShanStage.scala:49)
        at top.TopMain$.main(Top.scala:452)
        at top.TopMain.main(Top.scala)
1 targets failed
XiangShan.runMain subprocess failed
java.io.IOException: com.sun.jna.LastErrorException: [104] Connection reset by peer
        at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.doRead(UnixDomainSocket.java:151)
        at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.read(UnixDomainSocket.java:122)
        at mill.main.client.ProxyStreamPumper.run(ProxyStreamPumper.java:23)
        at java.lang.Thread.run(Thread.java:748)
Caused by: com.sun.jna.LastErrorException: [104] Connection reset by peer
        at org.scalasbt.ipcsocket.UnixDomainSocketLibrary.read(Native Method)
        at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.doRead(UnixDomainSocket.java:149)
        ... 3 more
```

zhaoyukoon, 2021-08-22:
i try to delete out folder and rerun, got following message:
```
java.io.IOException: com.sun.jna.LastErrorException: [104] Connection reset by peer
        at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.doRead(UnixDomainSocket.java:151)
        at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.read(UnixDomainSocket.java:122)
        at mill.main.client.ProxyStreamPumper.run(ProxyStreamPumper.java:23)
        at java.lang.Thread.run(Thread.java:748)
Caused by: com.sun.jna.LastErrorException: [104] Connection reset by peer
        at org.scalasbt.ipcsocket.UnixDomainSocketLibrary.read(Native Method)
        at org.scalasbt.ipcsocket.UnixDomainSocket$UnixDomainSocketInputStream.doRead(UnixDomainSocket.java:149)
        ... 3 more

```

poemonsense, 2021-08-22:
We don't test 4-core version regularly. It should be some parameterization-related issues. Perhaps you could make a pull request to do this?

fatalfeel, 2021-08-22:
check this 
https://github.com/OpenXiangShan/XiangShan/issues/931

closed 2022-03-16 by AugustusWillisWang