OpenXiangShan/XiangShan #948: 请教一个关于renametable的问题

daxxyy123, 2021-08-23:
renametable的io.flush来自roq的io.flushOut.valid信号，并在io.flushOut.valid的基础上延了一拍，这个时间roq已经被清空了，不应该再有指令提交，可是不知道为什么renametable的代码中，在io.flush时还有可能在io.archWritePorts有因为指令commit而产生的写入？
谢谢
when (io.flush) {
    spec_table := arch_table
    // spec table needs to be updated when flushPipe
    for (w <- io.archWritePorts) {
      when(w.wen){ spec_table(w.addr) := w.wdata }
    }
  }

YikeZhou, 2021-08-23:
阁下引用的代码是在 PR #202 添加的，此时 `flush` 信号尚未出于时序原因延迟一拍送到 rename 模块，`flush` 有效的当拍可能有一条指令写回。

daxxyy123, 2021-08-24:
谢谢，如您所说在rob 产生flush的当拍，确实可能还存在一条指令还需要提交，在这种情况下renametable是有可能有写操作的。但是现在CtrlBlock已经把rob的flush又延了一拍，renametable再收到flush的时候其实已经不会再有指令提交，所以现在其实是不会再有这种情况。不知我的理解是否正确？谢谢。

daxxyy123, 2021-08-30:
抱歉，我在公司内网，目前限于it安全策略，暂时没法对外提交代码。

closed 2021-11-18 by daxxyy123