OpenXiangShan/XiangShan #972: debug L2CacheTest.scala error in intellij

fatalfeel, 2021-08-28:
1.
requirement failed: Key XSCoreParamsKey is not defined in Parameters
java.lang.IllegalArgumentException: requirement failed: Key XSCoreParamsKey is not defined in Parameters
	at ... ()
	at chipsalliance.rocketchip.config$View.apply(Config.scala:28)
	at chipsalliance.rocketchip.config$View.apply(Config.scala:25)
	at xiangshan.HasXSParameter.$init$(Parameters.scala:174)
	at xiangshan.cache.DCache.<init>(DCacheWrapper.scala:132)
	at cache.L2TestTop.$anonfun$cores$1(L2CacheTest.scala:86)
	at scala.Array$.fill(Array.scala:270)
	at cache.L2TestTop.<init>(L2CacheTest.scala:86)
	at cache.L2TestTopWrapper.<init>(L2CacheTest.scala:260)
	at cache.L2CacheTest.$anonfun$new$57(L2CacheTest.scala:290)
	at ... ()
	at ... (Stack trace trimmed to user code only. Rerun with --full-stacktrace to see the full stack trace)

2. how do I debug /root/riscv_cpu/XiangShan/rocket-chip/src/main/scala/rocket/ICache.scala in intellij

ljwljwljwljw, 2021-08-29:
The code there has been deprecated and they will be removed form XiangShan. If you want to use TLC test, please take a look at https://github.com/OpenXiangShan/HuanCun/blob/main/src/test/scala/huancun/tlctest/RandomTester.scala. You can replace the dut with your own TLAdapterNode.

ljwljwljwljw, 2021-08-29:
To run unit tests in intellij:
1. Generate idea project configs by mill in command line
>  mill mill.scalalib.GenIdea/idea
2. Open the source file of unit test (eg: XXXTester.scala) in IDEA, there will be a green run button.

fatalfeel, 2021-08-29:
mill mill.scalalib.GenIdea/idea  already done before I post

AugustusWillisWang, 2023-02-15:
`L2CacheTest.scala` is not used for now. Closed.

closed 2023-02-15 by AugustusWillisWang