OpenXiangShan/XiangShan #995: XiangShan.test.runMain subprocess failed

abignail, 2021-09-03:
When I try to compile the "XiangShan-yanqihu"project by 'make verilog', I encounter a problem, as shown below. What is the problem, thanks.

Done elaborating.
1 targets failed
XiangShan.test.runMain subprocess failed
Makefile:46: recipe for target 'build/XSTop.v' failed
make: *** [build/XSTop.v] Error 1

poemonsense, 2021-09-03:
See

abignail, 2021-09-03:
Thank you poemonsense. This problem is hardware relate for me. I copy the Xiangshan project to my server, then it works well. The server in our group has enough memory for the project. The 'Done elaborating' takes about half an hour to finish, and the project uses about 45GB of memory in our server. I have to say, this project is very resource intensive.

poemonsense, 2021-09-03:
For minimal config, machine with 32GB memory is enough. But as you said, this is very resource intensive.

We are trying to minimize requirements to run XiangShan. We will update our tool chain once we have some progress.

closed 2023-02-15 by AugustusWillisWang