OpenXiangShan/XiangShan #997: latest master code "make help" fail ?

dongdeji, 2021-09-03:
root# make help
mill XiangShan.test.runMain top.SimTop --help
[205/265] fudian.compile
[info] compiling 1 Scala source to /share/dongdeji/xiangshan_work/xiangshan-master-20210903-mill/out/fudian/compile/dest/classes ...
[error] /share/dongdeji/xiangshan_work/xiangshan-master-20210903-mill/fudian/src/main/scala/fudian/Generator.scala:13:29: value getPackageName is not a member of Class[T]
[error]     val pkg = this.getClass.getPackageName
[error]                             ^
[error] one error found
1 targets failed
fudian.compile Compilation failed
Makefile:44: recipe for target 'help' failed
make: *** [help] Error 1

I change "fudian\src\main\scala\fudian\Generator.scala line13" to "val pkg = "PackageName"//this.getClass.getPackageName", then "make help" and "make verilog" cmds both goes fine.

poemonsense, 2021-09-04:
I cannot reproduce on my machine. Please try https://github.com/OpenXiangShan/XiangShan/issues/837 first

```
$ make help 
mill XiangShan.test.runMain top.SimTop --help                                                                                   
[265/265] XiangShan.test.runMain                           
Usage: xiangshan [options] [<arg>...]
                                                                                                                                
Shell Options                                                   
  <arg>...                 optional unbounded args                                                                              
  -td, --target-dir <directory>                                 
                           Work directory (default: '.')                                                                        
  -faf, --annotation-file <file>                                                                                                
                           An input annotation file
  -foaf, --output-annotation-file <file>            
                           An output annotation file                                                                            
  --show-registrations     print discovered registered libraries and transforms
  --help                   prints this usage text
Logging Options                                                 
  -ll, --log-level {error|warn|info|debug|trace}                                                                                
                           Set global logging verbosity (default: Warn    
  -cll, --class-log-level <FullClassName:{error|warn|info|debug|trace}>...
                           Set per-class logging verbosity
  --log-file <file>        Log to a file instead of STDOUT
  -lcn, --log-class-names  Show class names and log level in logging output
XiangShan Options                                               
  -dm, --disable-module-print <module>
                           The verilog 'printf' in the <module> and it's submodules will be removed

  -em, --enable-module-print <module>
                           The verilog 'printf' except the <module> and it's submodules will be removed

  -dall, --disable-all     All the verilog 'printf' will be removed

  --remove-assert          All the 'assert' will be removed

Chisel Front End Options                                        
  -chnrf, --no-run-firrtl  Do not run the FIRRTL compiler (generate FIRRTL IR from Chisel and exit)
  --full-stacktrace        Show full stack trace when an exception is thrown
  --chisel-output-file <file>                                   
                           Write Chisel-generated FIRRTL to this file (default: <circuit-main>.fir)
  --module <package>.<module>                                   
                           The name of a Chisel module to elaborate (module must be in the classpath)
FIRRTL Compiler Options                                         
  -i, --input-file <file>  An input FIRRTL file
  -o, --output-file <file>                                      
                           The output FIRRTL file
  --info-mode <ignore|use|gen|append>
                           Source file info handling mode (default: use)
  --firrtl-source <string>                                      
                           An input FIRRTL circuit string
  -fct, --custom-transforms <package>.<class>
                           Run these transforms during compilation
  --change-name-case <lower|upper>
                           Convert all FIRRTL names to a specific case
  -X, --compiler <none|high|middle|low|verilog|mverilog|sverilog>
                           The FIRRTL compiler to use (default: verilog)
  -E, --emit-circuit <chirrtl|high|middle|low|verilog|mverilog|sverilog>
                           Run the specified circuit emitter (all modules in one file)
......
```

dongdeji, 2021-09-04:
不是内存的问题，这个提示：
[error] val pkg = this.getClass.getPackageName
[error] ^

找不到getPackageName这个函数，你的机器上又是一切正常，那怀疑是不是java虚拟机版本相关，能否麻烦看下你的java虚拟机版本？linux shell上使用命令“java -version”
比如我的版本信息（比较老？）：
root# java -version
openjdk version "1.8.0_292"
OpenJDK Runtime Environment (build 1.8.0_292-8u292-b10-0ubuntu1~16.04.1-b10)
OpenJDK 64-Bit Server VM (build 25.292-b10, mixed mode)

poemonsense, 2021-09-04:
java 8确实编不过

我们推荐使用 graalvm，能减少内存占用，同时提高运行速度。https://www.graalvm.org/

dongdeji, 2021-09-04:
要java几以上可以呢？

poemonsense, 2021-09-04:
Java 11

closed 2022-03-16 by AugustusWillisWang