OpenXiangShan/XiangShan #999: "Bump chisel to 3.5 (#974)"标签之后注释掉了vlsi_mem_gen生成sram仿真文件的功能，为了FPGA验证的文件的完整性，应该将“--gen-mem-verilog full”参数配置到make verilog命令编译中，否者fpga验证缺少一堆sram模块，谢谢！

dongdeji, 2021-09-04:
MemLib Options
  -firw, --infer-rw        Enable read/write port inference for memories
  -frsq, --repl-seq-mem -c:<circuit>:-i:<file>:-o:<file>
                           Blackbox and emit a configuration file for each sequential memory
  -gmv, --gen-mem-verilog <blackbox|full>
                           Blackbox and emit a Verilog behavior model for each sequential memory

Makefile中line46修改成：
$(TOP_V): $(SCALA_FILE)
	mkdir -p $(@D)
	mill -i XiangShan.runMain $(FPGATOP) -td $(@D)                       \
		--config $(CONFIG) --full-stacktrace --output-file $(@F)     \
		--disable-all --remove-assert --gen-mem-verilog full \
		$(SIM_ARGS) \
		--num-cores $(NUM_CORES)

由于所在内网原因无法向外提交代码，所以只能提交issue反馈，谢谢！

poemonsense, 2021-09-04:
Thanks for reporting to us. It's a known issue. To be fixed by the transform described in issue #858.

closed 2021-09-04 by poemonsense

poemonsense, 2021-09-04:
Fixed by #1000