OpenXiangShan/XiangShan #1032: 关于lfst的问题

daxxyy123, 2021-09-14:
lfst查表的结果：
    if(StoreSetEnable) {//使用storeset
      // updatedUop(i).cf.loadWaitBit := lfst.io.lookup.rdata(i) // classic store set
      updatedUop(i).cf.loadWaitBit := lfst.io.lookup.rdata(i) && !isStore(i) // store set lite
      // updatedUop(i).cf.loadWaitBit := lfst.io.lookup.rdata(i) && io.fromRename(i).bits.cf.loadWaitBit && !isStore(i) // 2-bit store set
    } else {//使用waittable
      updatedUop(i).cf.loadWaitBit := io.fromRename(i).bits.cf.loadWaitBit && !isStore(i) // wait table does not require store to be delayed
    }
如果使用storeset我的理解load指令查询LFST，并根据SSID索引LFST找到相关store指令的sqIdx，然后应该要修改updatedUop(i).sqIdx，产生RS issue相关性，但是现在的代码里是直接使用io.enqLsq.resp(i).sqIdx，而且LFST的设计中其实也没有根据ssid查询sqIdxVec并把结果送出来只查了valid，好像没有设计完？？现在这个做法是一个很保守的结果应该是使用waittable时用的。这样其实storeset的作用就没有发挥出来。
不知道为什么这样设计？或是我的理解不对，谢谢。

AugustusWillisWang, 2021-09-14:
是的, 我们目前使用的是一个弱化版本的 Store Set. LFST 的结果目前使用的很保守, 没有实现原始论文中 Store Set 的全部功能.

与原始论文有差异的地方为:

> Making each of the stores dependent on the last fetched store within its store set. Each store specifies one dependence and each load specifies one dependence to form an in-order chain resulting in correct program behavior

目前代码中"追踪指令间的顺序并依次执行"的逻辑有待改进, 后面的指令要在前面的指令结束多个周期后才能开始执行. 这样的话, 如果我们遵循原论文中同一 Store Set 内指令定序的要求, 就会带来较大的性能开销.

我们正在更新 Store Set 以及相关机制的实现, 欢迎参与后续讨论!

daxxyy123, 2021-09-15:
好的，多谢

closed 2021-11-16 by daxxyy123