OpenXiangShan/XiangShan #1040: 请教roq中的intrEnable信号

daxxyy123, 2021-09-17:
在roq中intrEnable信号收hasNoSpecExec和loadstore指令的控制。
val intrEnable = intrBitSetReg && !hasNoSpecExec && !CommitType.isLoadStore(deqDispatchData.commitType)
当有非推测指令或是load/store指令时，intrEnable 关闭。
我的问题是：
1）为什么非推测指令需要关闭中断？指令集里好像没有这个要求？
2）load/store关闭中断的原因，代码注释里提了一下，意思是不是中断执行会导致当前被提交指令会在中断结束后再执行一次，而这对于MMIO是不允许的？
谢谢

poemonsense, 2021-09-17:
1) 我们现在CSR会在执行的时候就写入，发出中断，会导致中断触发，但这条指令实际已经执行却被标记了异常

2) MMIO和CSR有类似的，会在指令实际提交之前，就完成MMIO的读写，如果发生中断，同样会导致这条指令实际已经执行却被标记了异常

poemonsense, 2021-09-17:
这一块涉及到特权级的部分，因为不涉及到性能，但能够保证功能正确。我们的处理比较粗暴，上一次没有足够时间来进行细调，欢迎给我们提供优化思路

daxxyy123, 2021-09-17:
谢谢解释。按照现在的设计load/store指令在提交时物理地址和cacheable属性应该是知道的，否则load violation判断就有问题了吧，所以好像可以区分MMIO操作。
另外还想请教一下：
从微架构正确性角度考虑，当指令要提交时正好来了个中断，这时候应该继续提交指令还是执行中断？
比如假如中断正好是在提交load指令时到的，但是如果中断的结果会改变load指令的执行结果，比如中断处理程序会改变内存内容，而load指令只好要读这段内存，那此时从正确性角度考虑是应该先执行中断还是先执行load指令？（CSR写似乎也有这个问题）
不知道这种情况应该如何考虑，谢谢。

poemonsense, 2021-09-17:
被标记上中断的指令，其不应该被执行。所以如果CSR/访存指令被标记上了中断，那么这条指令不应该读写任何的CSR、不应该读写任何的内存

daxxyy123, 2021-09-18:
明白了，谢谢。那现在这种通过intrEnable实际延后中断处理的方式在将来的版本中需要调整把。

poemonsense, 2021-09-18:
这一块目前还没有开始调整优化的工作，接下来一个月应该会重新梳理这部分的代码，也欢迎给我们提建议

closed 2021-11-16 by daxxyy123