OpenXiangShan/XiangShan #1046: Verification flow

saw235, 2021-09-18:
Hi,

This is a very interesting project that showcases Chisel's capability. 
I am interested in understanding the verification flow this project is using. And how to go about integrating functional verification and random testing with Chisel.

Are there such documentations or slides available?

poemonsense, 2021-09-18:
We have some slides for RISC-V Summit China 2021 but they are currently in Chinese only. https://github.com/OpenXiangShan/XiangShan-doc

Our verification is mainly based on NEMU (an ISA simulator, or a RISC-V reference design like Spike) and difftest framework (https://github.com/OpenXiangShan/difftest). When XiangShan and NEMU have different behaviors, XiangShan runs into errors.

saw235, 2021-09-18:
Thank you for your response. 

I see. I have read the papers, but there is very little info on the QoR of the tape-out core. I'm just wondering if it encountered some synthesis/simulation mismatch issues in the actual chip, and if there are some clock/power gating strategies implemented as well.

poemonsense, 2021-09-18:
The chip is expected to return early 2021. Hope it works well

closed 2023-02-15 by AugustusWillisWang