OpenXiangShan/XiangShan #1118: 请教inclusive mshr中的probe_next_state

daxxyy123, 2021-10-15:
在inclusive/mshr中，有这样一段代码：
val probe_next_state = Mux(
    isT(meta.state) && req.param === toT,
    meta.state,//如果本节点当前是T，表明本节点原来是TIP或是Trunk，而client probe要求T，则应生长cache一致性树，本节点应该变成trunk，这样不能保持原状，否则对于TIP就不对了？
    Mux(meta.state =/= INVALID && req.param =/= toN, BRANCH, INVALID)//如果原先是INVALID同时现在要求toB，那meta状态似乎应该B而不是INVALID？？
  )
不太明白，能否帮忙解释一下，
谢谢

ljwljwljwljw, 2021-10-15:
probe的方向是manager -> client(例如L2 -> L1)，只会降低/维持client权限，不可能提高client的权限

daxxyy123, 2021-10-15:
多谢解释，假设拓扑如下：
L1          L1
   |          |
   |          |
  L2        L2
   |           |
        l3
例如这种情况：左边的l2原来是某个cacheline的TIP，收到了来自l3的probe toT请求（可能是由于右边的l2对下方l3发出acquire toT的请求），这时左边的l2似乎应该变成INVALID，l3变成TRUNK，而右边的l2变成TIP。
可是按照现在的代码左边的l2的meta状态还会保持原先的TIP？？
不知道我的理解是否正确，请帮忙看一下，谢谢

daxxyy123, 2021-10-15:
谢谢，是这样的我想错了，那请问什么样的情况会发出probe to T？

ljwljwljwljw, 2021-10-15:
> 谢谢，是这样的我想错了，那请问什么样的情况会发出probe to T？

这个跟具体应用场景相关了，有时候L3可能想确认一下L2是不是真的还有T权限(Directory可能记录的不准确)，就发出一个toT，然后L2如果真的有T权限，就返回TtoT，如果没有就返回NtoT/BtoT。

closed 2021-10-16 by ljwljwljwljw