OpenXiangShan/XiangShan #1194: MaskGen生成的mask错误

hhhhhyang, 2021-11-02:
我在测试小字节读写时使用到了merge方法，发现merge方法调用mask，mask调用MaskGen方法存在bug。
如果MaskGen的输入地址不是0的话，生成的mask是不正确的。
写下如下代码进行测试，
`val edgeOut_mask = edgeOut.mask(io.out.bits.address, io.out.bits.size)`
测试用例覆盖为，其中i为地址，s为size
`for{
    i <- 0 to 128
    s <- 0 to 6
  }`

1. 当地址为0时，波形如下图所示，生成的mask是正确的。
[image]

2. 当地址为1和2时，mask在size为0和1时错误
[image]

[image]

3. 当地址为4，5，6和7时，mask在size为0，1和2时错误
[image]

4. 当地址为8到F时，mask在size为0，1，2和3时错误。
5. 依此类推，当地址为0h10时，mask就在0，1，2，3，4错误。

大家有发现这个错误吗？

BigWhiteDog, 2021-11-02:
这并非错误，这个方法引用自tilelink源码中，给定对应地址时，生成对应size大小内的所有字节地址的mask，这个区间按size对齐，例如地址为1，size为1，那么生成0x1地址所在的2个字节的掩码，2字节应对齐到0x0，那么应该生成0b0011掩码。
更加详细的解释可以参阅网上的一篇文章https://www.cnblogs.com/wjcdx/p/10952169.html

hhhhhyang, 2021-11-02:
您好，这篇文章我也看过了。但还是存在疑惑，TL协议确实要求address和size对齐，但是这无法解释MaskGen当addr为1，size为0(即byteSize为1字节)时，MaskGen的输出是0b0010。按照我的理解，任意地址都会和1字节对齐，那么MaskGen的输出应该是0b0001。或者说，地址为2时，size为0或者size为1都可以对齐，但是MaskGen的输出依然不是0b0001和0b0011.

BigWhiteDog, 2021-11-02:
请仔细看TileLink协议的Byte Lanes一节，并配合规范定义的图示理解

hhhhhyang, 2021-11-02:
感谢！现在我明白了~

closed 2021-11-02 by hhhhhyang