OpenXiangShan/XiangShan #1204: 请问关于FTQ中的fallThroughError

hyf6661669, 2021-11-07:
https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/frontend/NewFtq.scala#L87-L89
```
  def fallThroughError() = {
    !carry && startAddr(instOffsetBits+log2Ceil(PredictWidth), instOffsetBits) > pftAddr
  }
```
上电复位之后, FTB是空的，所以BPU对于FTB的访问一定是miss的, 此时FTB里会用如下代码生成一个pftAddr:
```
  def onNotHit(pc: UInt) = {
    pftAddr := pc(instOffsetBits + log2Ceil(PredictWidth), instOffsetBits) ^ (1 << log2Ceil(PredictWidth)).U
    carry := pc(instOffsetBits + log2Ceil(PredictWidth)).asBool
    oversize := false.B
  }
```

按照PredictWidth = 16, instOffsetBits = 1的设置, 那么有:
```
pftAddr = {~pc[5], pc[4:1]}
carry = pc[5]
```
那么这个fallThroughError()什么时候会算出1呢, 暂时没想明白? 我看FTB的完整的tag应该是有26-bit, 但是实际只存储了20-bit的tag(应该是为了省面积), 所以会出现false_hit的情况，那么这个fallThroughError()应该是用来检测false_hit的么?

Lingrui98, 2021-11-08:
你说的是正确的，这个函数是为了避免false_hit时算出错误的fallThroughAddr，造成IFU取指错误，因为目前IFU的取指逻辑需要FTQ保证fallThrough都是预期之内的值。另外，这个Bundle里定义的这个函数目前没有使用，真正使用的是这里https://github.com/OpenXiangShan/XiangShan/blob/f5933401f63c2b3a10f9b523a19dbe587b0c303f/src/main/scala/xiangshan/frontend/FrontendBundle.scala#L35-L44 的这个函数，在给IFU发取指请求时调用，它会保证fallThroughAddr会在(startAddr, startAddr+FetchBytes+2]这个范围之内。

hyf6661669, 2021-11-08:
请问区间尾部为何是FetchBytes+2而不是FetchBytes呢..

Lingrui98, 2021-11-08:
如果是FetchBytes，如果最后两byte是跳转指令的话，IFU需要多发一次请求取到对应的指令码，这个逻辑应该不是必要的，后续可以进行优化

closed 2021-11-08 by hyf6661669