OpenXiangShan/XiangShan #1208: A proposal of uploading verilog files

hyf6661669, 2021-11-09:
Since it requires lots of memory to generate verilog from big chisel projects like XiangShan, I hope you could also upload the verilog files generated from the latest chisel src (or from a stable version). So beginners without a powerful computer (It is sad that my computer only has 8GB RAM) could also use verilog files to do some simulation work.

poemonsense, 2021-11-09:
We don't recommend to track verilog files via git since they are not a part of the source code. Besides, we used to upload Verilog files along with our stable releases in https://github.com/OpenXiangShan/XiangShan/tags but noticed that these verilog files only represented only fixed parameters for XiangShan. This is not what we want from XiangShan.

However, as you want the Verilog files for simulation, I've uploaded the generated verilog files for single-core XiangShan with default configs. [xiangshan_29230e8.tar.gz](https://github.com/OpenXiangShan/XiangShan/files/7502626/xiangshan_29230e8.tar.gz)

Besides, we do want to see more people joining us with their laptops. However, as stated in https://github.com/OpenXiangShan/XiangShan/issues/852#issuecomment-878716629, we prefer to solve the problem by
- offering small designs (MinimalConfig of XiangShan, in-order core NutShell, etc)
- offering partial designs by using fake logics in other unrelated modules (FakeFrontEnd, FakeCache, etc)
- offering public cloud-based development environments
- optimizing Chisel, verilator, and other tools.

The serious fact is that XiangShan is large and will definitely become larger in the future. We don't want to limit XiangShan to small configurations that fits into laptops, but try to find a way to develop large designs with limited resources (and uploading verilog files does not address this issue).

hyf6661669, 2021-11-09:
> We don't recommend to track verilog files via git since they are not a part of the source code. Besides, we used to upload Verilog files along with our stable releases in https://github.com/OpenXiangShan/XiangShan/tags but noticed that these verilog files only represented only fixed parameters for XiangShan. This is not what we want from XiangShan.
> 
> However, as you want the Verilog files for simulation, I've uploaded the generated verilog files for single-core XiangShan with default configs. [xiangshan_29230e8.tar.gz](https://github.com/OpenXiangShan/XiangShan/files/7502626/xiangshan_29230e8.tar.gz)
> 
> Besides, we do want to see more people joining us with their laptops. However, as stated in [#852 (comment)](https://github.com/OpenXiangShan/XiangShan/issues/852#issuecomment-878716629), we prefer to solve the problem by
> 
> * offering small designs (MinimalConfig of XiangShan, in-order core NutShell, etc)
> * offering partial designs by using fake logics in other unrelated modules (FakeFrontEnd, FakeCache, etc)
> * offering public cloud-based development environments
> * optimizing Chisel, verilator, and other tools.
> 
> The serious fact is that XiangShan is large and will definitely become larger in the future. We don't want to limit XiangShan to small configurations that fits into laptops, but try to find a way to develop large designs with limited resources (and uploading verilog files does not address this issue).

Really thanks for sharing your ideas about lowering the requirements of exploring XiangShan. I think the MinimalConfig option is very good and I can still learn some basic ideas about OoO design.

closed 2021-11-09 by hyf6661669