OpenXiangShan/XiangShan #1220: 请教noninclusive mshr中的preferCache信号。

daxxyy123, 2021-11-12:
在noninclusive mshr中：
val preferCache = req.preferCache || cache_alias
请问preferCache和cache_alias分别代表什么含义？
谢谢

ljwljwljwljw, 2021-11-12:
`preferCache`和`alias`都是由client向cache发送请求时带的tilelink user bit
- `preferCache`是client用来控制下层cache自己要不要存储请求对应块的, 如果存,那么cache与client形成inclusive关系,如果不存,cache与client形成exclusive关系
- `alias`是cache别名信息,由于香山的L1Cache为128KB 8Way,而页大小仍然是4KB, cache中14位的index超过了12位的页号,在L1中可能存在cache别名问题,目前香山通过在L2中存储L1 client 14-12 = 2 bit的虚拟地址来检测别名,`cache_alias`为`true`代表着L2检测出了L1的别名冲突,此时L2会向L1发送probe将冲突向probe掉.

daxxyy123, 2021-11-15:
多谢，关于alias想在多问几句。
不知道香山处理alias问题的思路，在l1 dcache中似乎也应该做相关处理，但是我看到最新的代码中LoadPipe没有做alias的处理，只是在missqueue和refillpipe中有些处理，是否还没有合入？
还有就是cache_alias 的判断为什么只针对发出req的cache？其它client也有可能和当前req的cacheline发生alias？请求的cacheline在发出req的cache中命中，那就应该可以本地hit，为什么还要向下发出acquire？：
iam := OHToUInt(getClientBitOH(io.alloc.bits.source))
val req_client_meta = clients_meta(iam)
val cache_alias = req_client_meta.hit && req_acquire &&
    req_client_meta.alias.getOrElse(0.U) =/= req.alias.getOrElse(0.U)

ljwljwljwljw, 2021-11-17:
## 香山对cache alias的处理

`l1`向`l2` `acquire block A`的时候，会带2位的虚地址下来通过`tilelink user bit`，`l2`会把这2位虚地址保存起来(在`client dir`中)，下次`l1`再`acquire block B`的时候，如果`l2`中记录`l1`信息的`client dir` hit，`l2`会把新的2位和之前存的两位做对比，如果不一样，就说明`l1`以为自己miss了，但其实是别名了，`block A`和`block B`其实是虚地址不同物理地址相同的同一个块，且`block A`已经在`l1`中了，此时`l2`会向上发出`probe`，目标地址为`block A`和`block B`的物理地址，并在B通道的`data`字段中带上`block A`的2位虚地址，`l1`收到该`probe`后，会使用`block A`的2位虚地址替换`probe`请求原本的物理地址中的对应2位(12, 11)，并使用替换后的地址去完成`probe`操作，最终给`l2`一个`probe ack`。上述过程结束后，`l1`中的`block A`就被`probe`掉了，`l2`再向`l1`返回`block B`，从而保证物理地址相同的块在`l1`中只会存在于1个位置，避免别名问题。

ljwljwljwljw, 2021-11-17:
> 不知道香山处理alias问题的思路，在l1 dcache中似乎也应该做相关处理，但是我看到最新的代码中LoadPipe没有做alias的处理，只是在missqueue和refillpipe中有些处理，是否还没有合入？

- 基于我们的处理方式，l1 cache不需要对LoadPipe做特殊处理，仍然拿虚地址去索引，读data即可

> 还有就是cache_alias 的判断为什么只针对发出req的cache？其它client也有可能和当前req的cacheline发生alias？

- l2中的块都是用物理地址管理的，不存在别名问题，只要保证l1的每个client内部不会出现别名即可

> 请求的cacheline在发出req的cache中命中，那就应该可以本地hit，为什么还要向下发出acquire？

- `req_acquire`的含义是当前mshr处理的请求为acquire请求，并不是说当前mshr需要schdule一次acquire

daxxyy123, 2021-11-18:
感谢解释。

closed 2021-11-18 by daxxyy123