OpenXiangShan/XiangShan #1231: 关于FallThruError的问题

ddxoxpp, 2021-11-15:
https://github.com/OpenXiangShan/XiangShan/blob/96b1e495b767b89ccbc372083ebd790aee0a36fb/src/main/scala/xiangshan/frontend/PreDecode.scala#L217
predecoder中直接把FallThruError作为mispred的条件之一，从而导致产生ifu redirect flush。如果预测时没有用到FallThruAddr作为nextPC，这里直接flush岂不是不合理；或者既然带FallThruError的req一定会在ifu f3 产生redirect flush，为何不在更早的时候检查FallThruError，比如说bpu预测阶段hit了ftb entry就检查，更早进行纠正？

Lingrui98, 2021-11-15:
目前在ftq向ifu发送请求的时候检测FallThruError，留到f3再flush的原因是需要利用ifu的预译码信息，同时也简化重定向通路。
以目前的FTB容量（4K项），FallThruError极少发生，它不是性能损失的重点

ddxoxpp, 2021-11-15:
还有个问题就是为啥不像bpu预测时那样直接比较target和ghist来判断是否需要ifu redirect，而是通过上述六种fault比较，那样时序更优吗？上述六种fault是完备的吗，比如说我看只检查了return没有完全检查jalr。

Lingrui98, 2021-11-15:
ifu只判断bpu判断不了的而在有指令预译码信息后能判断的误预测，你提到的jalr误预测是不能检查的，因为target需要寄存器的值产生

ddxoxpp, 2021-11-15:
谢谢解答！

closed 2021-11-16 by ddxoxpp