OpenXiangShan/XiangShan #1232: 请教noninclusive mshr中的transmit_from_other_client信号

daxxyy123, 2021-11-15:
在noninclusive mshr中的transmit_from_other_client：
val transmit_from_other_client = !self_meta.hit && VecInit(clients_meta.zipWithIndex.map {
    case (meta, i) =>
      (req.opcode === Get || i.U =/= iam) && meta.hit // ？？
  }).asUInt.orR
当i.U === iam时，meta.hit，req.opcode === Get，这是什么场景？好像很奇怪。
谢谢。

wakafa1, 2021-11-15:
我们目前暂时没有这样的场景，但是同一个 Client 既发 Acquire 又发 Get 是合法的操作

daxxyy123, 2021-11-16:
谢谢，不过我还是不太明白：这个client既然自己可以meta hit为什么还要向下发出get请求？另外既然它自己可以hit了，为什么还要约束只有它自己发出的是get操作，才能再次从它自己那里transmit？

wakafa1, 2021-11-16:
对一般的 Client 而言，自己 meta hit 还要发出 get 是很不寻常的操作，我们目前也不知道什么场景会用到，但它是符合手册规范的。但是如果自己 meta hit 但发出的还是 acquire，那么就不符合手册规范了，所以我们有这样的约束

daxxyy123, 2021-11-16:
多谢回复，那既然get可以，似乎put、hint、atomics也应该可以，只需要排除acquire就可以了？

wakafa1, 2021-11-17:
是的，hint 和 put 我们有另外的处理，而原子访存我们是在 D$ 里面处理的，因此收不到 atomics 的 request

daxxyy123, 2021-11-18:
多谢解释

closed 2021-11-18 by daxxyy123