OpenXiangShan/XiangShan #1236: 关于secondary miss的问题

ddxoxpp, 2021-11-16:
https://github.com/OpenXiangShan/XiangShan/blob/08b0ab9f4e07e932e54b05a71a82592bc4dd2599/src/main/scala/xiangshan/frontend/IFU.scala#L468-L471
f2 和f1 如果具有secondary miss关系，那么是否应该反应到f1的对应bank hit信号上去：
https://github.com/OpenXiangShan/XiangShan/blob/08b0ab9f4e07e932e54b05a71a82592bc4dd2599/src/main/scala/xiangshan/frontend/IFU.scala#L233
因为该f1 req进入f2 后miss bank直接使用  ```val reservedRefillData = Reg(Vec(2, UInt(blockBits.W)))```寄存器的值，再去执行一遍f2的状态机取miss bank浪费了时间。
https://github.com/OpenXiangShan/XiangShan/blob/08b0ab9f4e07e932e54b05a71a82592bc4dd2599/src/main/scala/xiangshan/frontend/IFU.scala#L437-L440
然后相应的修改f2_data逻辑为：
```  f2_datas.zipWithIndex.map{case(bank,i) =>
    if(i == 0) bank := Mux(f2_bank_hit(i),Mux(sec_miss_reg(2),reservedRefillData(1),Mux(sec_miss_reg(0),reservedRefillData(0),f2_hit_datas(i))),f2_mq_datas(i))
    else bank := Mux(f2_bank_hit(i),Mux(sec_miss_reg(3),reservedRefillData(1),Mux(sec_miss_reg(1),reservedRefillData(0),f2_hit_datas(i))),f2_mq_datas(i))
  }```
不知道我理解的对不对。

jinyue110, 2021-11-17:
您好，感谢反馈，这个问题已经在 #1228 中修复了
[https://github.com/OpenXiangShan/XiangShan/pull/1228/commits/2a76f5b7ab8705f2ca841ee57a6d9a1c92bcd81a](url)

ddxoxpp, 2021-11-17:
谢谢解答

closed 2021-11-18 by ddxoxpp