OpenXiangShan/XiangShan #1240: 请教关于noninclusive mshr中的replace_param参数的问题

daxxyy123, 2021-11-17:
noninclusive mshr中的replace_param通过如下代码得到，
val replace_param = MuxLookup(
    Cat(self_meta.state, replace_clients_perm),
    TtoB,
    Seq(
      Cat(BRANCH, INVALID) -> BtoN,
      Cat(BRANCH, BRANCH) -> BtoB,
      Cat(BRANCH, TRUNK) -> TtoT,  //本地是branch，client怎么可能是trunk？？
      Cat(BRANCH, TIP) -> TtoT,    //??
      Cat(TIP, INVALID) -> TtoN,
      Cat(TIP, BRANCH) -> TtoB,
      Cat(TRUNK, TIP) -> TtoT
    )
我的问题是既然self_meta=B，它上方的client为什么会是trunk或tip，不应该出现这种情况呀？trunk和tip应该只能出现在branch的下方，但是代码中为什么会有 Cat(BRANCH, TRUNK) 和Cat(BRANCH, TIP)的情况，而且replace参数是TtoT？
能否请帮忙解释一下，谢谢。

linjuanZ, 2021-11-17:
这里的两种情况确实是冗余的，在我们的设计里应该是不会出现的。我们会尽快修改。感谢指出！

replace_paraｍ 的这些情况当初是这么考虑的：self_meta 有 3 种权限（B, T, TT），replace_clients_perm 有 4 种权限（N, B, T, TT），笛卡尔积之后有 12 种组合，划掉不可能出现的组合剩下了这 7 种情况。因为是 non-inclusive，我们认为 L3 及 L3 以上的权限 >= L2 及 L2 以上的权限就是合法的，所以如果 L3 是 B，L2 及以上是 T，那么 L3 及 L3 以上权限就是 T，所以不违反我们的定义。

daxxyy123, 2021-11-17:
多谢解释！

closed 2021-11-17 by daxxyy123