OpenXiangShan/XiangShan #1244: 关于noninclusive mshr中的probe_next_state的问题。

daxxyy123, 2021-11-17:
noninclusive mshr中处理收到b req后的meta变化的流程如下：
def onBReq(): Unit = { 
    /*
      if this request is from probe helper, the req param must be toN,
      but self state should not change to INVALID
      --------------------------
      self state   |  after probe
      --------------------------
        TIP       ->    TIP
        TRUNK     ->    TIP
        BRANCH    ->    BRANCH
      --------------------------
    */
    new_self_meta.dirty := self_meta.hit && self_meta.dirty || probe_dirty //probe_dirty:发出的probe req返回data
    new_self_meta.state := Mux(self_meta.hit,
      Mux(req.fromProbeHelper && !probeAckDataThrough, 
        Mux(isT(self_meta.state), TIP, BRANCH),
        probe_next_state(self_meta.state, req.param)
      ),
      self_meta.state 
    )

其中如果正常probe且hit的情况下new_self_meta.state=probe_next_state(self_meta.state, req.param)
在probe_next_state中：
def probe_next_state(state: UInt, param: UInt): UInt = Mux(
    isT(state) && param === toT,  //如果state是trunk？？--probe只能降权限，所以state只能是tip
    state,
    Mux(state =/= INVALID && param =/= toN, BRANCH, INVALID)
  )
我的问题是比如以下这种拓扑：
 nodeA（tip）
 nodeB（trunk）
 nodeC （trunk）
假如nodeC向nodeB发出probe toT（我的理解toT应该是to tip不是to trunk），这样本来应该nodeB应该升权限到tip，nodeA应该降权限到none。
但是按照现在的probe_next_state判断，因为isT在tip和trunk时都为高，这样实际上nodeB还是trunk，nodeA也仍然是tip，这样似乎不大对。不知道是否是我的理解不对，能否请帮忙解释一下，谢谢。

ljwljwljwljw, 2021-11-19:
> 但是按照现在的probe_next_state判断，因为isT在tip和trunk时都为高，这样实际上nodeB还是trunk，nodeA也仍然是tip，这样似乎不大对。不知道是否是我的理解不对，能否请帮忙解释一下，谢谢。

按照现在的处理逻辑，从结果上看，nodeB仍然是Trunk，这符合Probe toT的参数要求，如果想实现您所说的将toT精确到toTip，可能需要扩展tilelink协议，添加这样的参数，或者通过user bit来提供更多的信息。目前香山真实系统种不会出现probe toT的请求，如果考虑通用性的话，我认为保持trunk不变的好处在于可以减少总线上的请求数量，例如nodeB收到toT后发现自己已经满足，便不在继续向上层发probe。

ljwljwljwljw, 2021-11-19:
https://github.com/OpenXiangShan/HuanCun/blob/1f8ab4199ca3e35ca25aaba2c1a831e0ce17716f/src/main/scala/huancun/noninclusive/MSHR.scala#L621-L633

目前我们的b_schedule其实并没有考虑到toT的情况

daxxyy123, 2021-11-19:
感谢您的耐心解释，我觉得在tilelink协议里参数定义中似乎对于“T“”比较含混，比如BtoT、TtoB，那T就应该是指tip；但是TtoT、toT、TtoN等情况仔细理解似乎就有歧义,好像trunk或tip都可以，比如toT，到底是应该指tip还是trunk不是很清晰，这一块我一直不是很明白，tilelink协议好像要故意把trunk和tip混在一起，但是这两者的权限差异其实又非常大，不知道您这边是如何理解的？谢谢

ljwljwljwljw, 2021-11-21:
我认为不区分`Tip`和`Trunk`和tilelink的树形结构有关，由于手册规定`Tip`必须生长在`Trunk`上，以上面的nodeA, nodeB, nodeC的层次关系为例，假设我们在参数中严格区分`Tip`和`Trunk`，若nodeB向先nodeC acquire `Tip`权限，则nodeB变为`Tip`，nodeC变为`Trunk`，此时如果nodeA再向nodeB acquire `Tip`权限，则nodeA变为`Tip`，nodeB和nodeC都成为`Trunk`，然而此时nodeC无法得知其上方的nodeB已经由`Tip`变为`Trunk`了(nodeB由`Tip`变`Trunk`时并不会向下通知nodeC)。从这个例子可以看出nodeC仅能得知其上方所有节点的整体权限是T，但不能精确的知道nodeB是`Tip`还是`Trunk`。

正是由于`Tip`会在一些情况下向`Trunk`转变而不需要通知下方节点，所以下方节点只能记录其上方全部节点的整体权限，而无法得知上方某个具体节点的权限，不过这并不影响绝大多数常见的一致性场景，并且简化了参数设计，减少了总线上的消息数量，因此tilelink将`Trunk`和`Tip`混合在了一起(猜测)。或许像Probe toT、report TtoT这样的存在歧义的场景，应该由实现者决定具体的应对方式，不必因为这样不常用的场景增加协议复杂度？

感谢您的提问，和您讨论让我们重新思考和学习了很多，欢迎您继续跟我们交流！

daxxyy123, 2021-11-22:
多谢，这个解释很合理，对于下方节点更有意义的是上方整体权限，而且只要再增加一种状态，param的设计就会多出许多种情况，其实根据本节点状态确实还是基本可以推理出T的含义，从压缩消息长度的角度说这样确实很合理。
感谢您不厌其烦的解答，贵团队的开诚布公和热情让我个人受益良多，非常感谢能有机会向这样一个优秀的团队学习！

closed 2021-11-24 by daxxyy123