OpenXiangShan/XiangShan #1250: 关于noninclusive mshr中 onAReq产生new_self_meta.clientStates代码的问题

daxxyy123, 2021-11-18:
在onAReq中 new_self_meta.clientStates由如下代码产生，
def onAReq(): Unit = {
......

new_self_meta.clientStates.zipWithIndex.foreach {
      case (state, i) =>
        when(iam === i.U) {
          state := Mux(req_acquire,
            Mux(req_needT || req_promoteT, TIP, BRANCH),
            Mux(
              req.opcode === Get,
              Mux(clients_meta(i).hit, BRANCH, INVALID)
              Mux(clients_meta(i).hit, clients_meta(i).state, INVALID)//这对应req.opcode ===put/hint的情况，client（iam）原来的状态 
                     //可以是T或B，比如是tip，这时l2需要通过probetoN获取tip权限才能支持put/hint操作,对应client（iam）的状态 
                    //应该会变成invalid,这样new_self_meta.clientStates应该是invalid而不是不变才对呀？？
            )
          )
        }.otherwise {
          state := Mux(
            req_acquire,
            Mux(
              req.param =/= NtoB || req_promoteT,
              INVALID,//本地节点将为T，所以client都是invalid
              Mux(clients_meta(i).hit, BRANCH, INVALID)
            ),
            Mux(
              req.opcode === Get,
              Mux(clients_meta(i).hit, BRANCH, INVALID),//本节点需要变成tip，client只能变成branch
              Mux(clients_meta(i).hit, clients_meta(i).state, INVALID)//同iam===i.U的问题？？
            )
          )
        }
    }
......
}
另外，请问为什么 new_self_meta.clientStates和new_clients_meta的产生方法并不是完全一样？这两者描述的clientstate为什么有差异？
谢谢

ljwljwljwljw, 2021-11-23:
- 关于Put的疑问请 @wakafa1 为您解答
-  关于self dir和client dir的问题请 @linjuanZ 为您解释

linjuanZ, 2021-11-23:
new_self_meta.clientStates 和 new_clients_meta 的逻辑确实是完全一样的，只是 onAReq() 里面写法稍有不同，在 new_clients_meta 里面 Get 的处理放在了后面，本质上是一样的。

wakafa1, 2021-11-25:
您的代码可能不是最新的，在 iam =/= i.U 的时候，我们额外对 req_put 做了判断：
https://github.com/OpenXiangShan/HuanCun/blob/7048298af7f66e5e20ab1512edcc1f88f5dd2520/src/main/scala/huancun/noninclusive/MSHR.scala#L342
另外在我们目前的设定下，只有一些外设会发 Put 请求，我们不会在 Dir 里维护它的状态，所以不会出现 iam === i.U 的情况

daxxyy123, 2021-11-25:
多谢解释。

closed 2021-11-25 by daxxyy123